robotology/mesh-support #589: Update firmware on STRAIN2 boards

fiorisi, 2017-12-27:
The robot is not starting. Probably there is an issue with the firmware version of the strain2 boards that have been recently [replaced](https://github.com/robotology/icub-support/issues/588).
In particular, might be necessary to update it from the version 1.0 to 2.0. 
A candidate could be the same version that is running on the r_leg FT sensor: [2.0.27](https://github.com/robotology/icub-firmware-build/blob/devel/CAN/strain2/strain2-v2.0.27.gain26.6.hex)
@marcoaccame @valegagge can you help us to update the firmware?

See attached log (line "1351 9,028682") and screenshot for further information.
**LOG**
[log_icub-head_yarprobotinterface_1684.txt](https://github.com/robotology/icub-support/files/1589631/log_icub-head_yarprobotinterface_1684.txt)

**FIRMWARE VERSION OF THE NOT WORKING FT IN THE LEFT LEG**
[image]

**WORKING FT IN THE RIGHT LEG**
[image]

traversaro, 2017-12-27:
If I remember correctly, there was a way to have the yarprobotinterface ignore this version checks by putting the desired version of `0.0.0` in some configuration files of the robot. @fjandrad can probably point you to the correct configuration files.

fjandrad, 2017-12-28:
We modified locally the files of the right leg iCubGenova04 hardware FT in [here](https://github.com/robotology/robots-configuration/blob/791e4d7453300bbb2496fd8d79a928975f99b22d/iCubGenova04/hardware/FT/right_leg-eb8-j0_3-strain.xml#L22). Changing the protocol major to 0 minor was already at 0. This was mentioned to be a **temporary solution** since what it does is ignore some checks.  This can be done for all sensors.

fiorisi, 2017-12-28:
Thanks @traversaro and @ fjandrad, we can try that, but don't you think that we should update the firmware to the same version of the right leg STRAIN2 ft?

traversaro, 2017-12-28:
Yes, I think that actually both thing are necessary (updating the firmware and putting the 0.0.0 version in the configuration files).

fiorisi, 2017-12-28:
Thanks @traversaro and @fjandrad , with the help of @randaz81 we managed to have all the FT working!

closed 2017-12-28 by fiorisi