robotology/mesh-support #590: Hardware fault on neck-yaw

diegoferigo, 2018-01-05:
### Description of the failure

The `neck-yaw` joint goes often in hardware fault.

### Detailed conditions and logs of the failure

```
...
24226	7676,522220	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 593s 971m 749u: (code 0x02000002, par16 0x0000 par64 0x0000000000000000) -> MC: i2t limit exceeded. The motor has been turned off to prevent it from being damaged by overheating due to a continuous high current. par16 = ID of joint. + .
...
8617	2623,342583	ERROR	Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=0) for BOARD head-eb20-j0_1 IP 10.0.1.20
8618	2623,645545	ERROR	embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD head-eb20-j0_1 IP 10.0.1.20 for nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target
8619	2623,645608	ERROR	embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD head-eb20-j0_1 IP 10.0.1.20 joint  0
```
cc @marcoaccame @ale-git

julijenv, 2018-01-05:
Hi @diegoferigo ,
Just to let you know that I noticed that the cable of the yaw joint is not position the best way possible and could affect the functionnality of this joint in the near future. When I will have time to access it I will reposiiton it.

marcoaccame, 2018-01-05:
About `i2t limit exceeded`: we have slightly increased the value of param `motorPeakCurrents`. Pls let us know if it happens again.

About: `embObjMotionControl::getRefVelocityRaw() could not read ...`: that typically happens when a board does not work properly... strange. Do you have the whole log?

gabrielenava, 2018-01-05:
> About: embObjMotionControl::getRefVelocityRaw() could not read ...: that typically happens when a board does not work properly... strange. Do you have the whole log?

I'm not completely sure, but if I remeber correctly a similar error is printed in the logger every time the home position is setted using yarpmotogui. Seems to be in someway related to this known issue: https://github.com/robotology/icub-main/issues/445

diegoferigo, 2018-01-05:
@julijenv Awesome, thanks! Also @arrenglover noticed several time this behavior in the past.

@marcoaccame After your edits, the first runs of the YOGA demo for once or twice triggered a hardware fault, but after for more than 10 runs the problem didn't occur again. On Monday I will perform other tests (@claudia-lat as well) and I will keep you updated.

> Do you have the whole log?

I don't have that specific log anymore, but I think the situation is as @gabrielenava described. I remember I read an error similar to the one posted in that issue in the terminal (e.g. of the yarpmotorgui). Probably the error in the logger is linked somehow to that one.

julijenv, 2018-01-16:
@diegoferigo , any feedback, can we close it?

arrenglover, 2018-01-16:
I still had a hardware fault on the neck yaw on Friday during the demo.

diegoferigo, 2018-03-30:
During the preparation of the last demo we never faced this issue anymore. I'm not aware if further maintenance has been done since last comments of this issue. I think we can close it for the time being.

closed 2018-03-30 by diegoferigo