robotology/mesh-support #591: eye tilt is unstable below 0 degrees

arrenglover, 2018-01-12:
### Description of the failure
The eye tilt motor becomes unstable when command to a state less than 0 degrees. The eyes oscillate between the commanded position and the maximum tilt value 

### Detailed conditions and logs of the failure

The failure could be replicated using the yarpmotorgui to command the eye position

julijenv, 2018-01-16:
Hi @arrenglover ,
whenever next time you go to the robot try to fetch me and we will try to figure out what happens.

gabrielenava, 2018-01-23:
This happened also to me while testing the yoga demo yesterday (but I was not moving the eyes), I put the eyes tilt in idle through the yarpmotorgui.

arrenglover, 2018-01-23:
Sorry @julijenv I didn't manage to try to debug this last week. I understand you aren't here this week and it seems the problem is now much worse - we cannot use the eyes of the robot at all at the moment. Can you please mention someone else that can help to fix this issue?

arrenglover, 2018-01-23:
@julijenv - we have the robot booked all day tomorrow. I hope we can have a look at the issue ASAP in the morning. thanks.

julijenv, 2018-01-23:
Hi @arrenglover , Ill be there tomorrow morning without a doubt Ill get a neye even before you do.

arrenglover, 2018-01-24:
Fixed - thanks @julijenv. The issue was mechanical - a loose screw connected to the tilt motor.

closed 2018-01-24 by arrenglover