robotology/mesh-support #594: iCub collapses after a few minutes of use

ems7, 2018-01-26:
Hi,

The Edinburgh iCub (with talking head) keeps collapsing after a few minutes of being switched on.

This has been occurring since just after Julien visited in December. No changes were made to the robot since Juliens visit, so potentially something went wrong during the firmware update.

The issue is possibly a driver issue, for communication between the pc104 and the motor control boards. However, it should be noted that this issue occurs whether the motors are in use or not.

logs from the robot can be found here:
syslog:
https://drive.google.com/file/d/1unoN8UUsAndsKfg4HKc4U7vvk0NtXQI8/view

startuplog:
https://drive.google.com/file/d/1otBgg1Ew0rc5gPx-jEsqhQJFrfZyVvB_/view

log_pc104_yarprobotinterface_1481:
https://drive.google.com/file/d/1JEVgpU-CX6Mco6LszMujz6jXQg0Yztmu/view

 log_pc104_imuFilter_1688
https://drive.google.com/file/d/1nHxV7FdEnCRJA8zy5noCE6Psfe3AzZHe/view

log_icubsrv_iKinGazeCtrl_13713
https://drive.google.com/file/d/16WFr223fijQ3dI0g8I1Iw6NHGJ3ZWAMU/view

lastlog
https://drive.google.com/file/d/1SK6IQGKpR56CFVqgxU8h2EziEC8REBcz/view

faillog
https://drive.google.com/file/d/15GWQtupHzJ2KofctXoDiXwWIIj-CURrD/view

julijenv, 2018-01-31:
Hi @ems7,

1. syslog:
the syslog is full of error due to unreckonized usb driver, module or hardware as well as the one of the firewire - for this latter , could you please tell us if the cameras are still working ??? those hardware are the only one using firewire connections

2. startuplog:
can't see anything strange -> what is the behaviour?!??? what was the problem during the run?
log_pc104_yarprobotinterface_1481:
from the log I can see tje you faulted the robt right from the start nad finally after (i dont know exctly but) a while, the last message is left_Arm AEA_encoder error, which is weird if the robot didnt even move (did you move the left arm by any chance to try to trigger this event on purpose - which will be nice by the way) let me know
3. log_pc104_imuFilter_1688
looks ok to me, but of course am not an expert
4. log_icubsrv_iKinGazeCtrl_13713
no errors popping up I guess it is ok as previously said Im not an expert
5. lastlog:
286kb of nothing inside :-S
6. faillog
empty as well :-S

To sum up, I would love to have the point of view of some software guys to check the syslog to understand what could be wrong inside the head to make those erro message for the usb port ad firewire (of course if you guys from Edinburgh can give me feedback aout the possibility that the one of both of the cameras are not working, it would be great).

Secondly, I need you to move the left arm in every single position possible by hand by leaving the 3 first motors in IDLE or by actuated it in order to detect if the error with the AEA encoder is repeatable and define or is it due to over time that the error pop up.

> <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 924s 563m 891u: (code 0x02000008, par16 0x0002 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

If it is repeatable surely you will need to check the wire connection on the sensor:
[image]
The cable gets out of the top of the upper arm (see picture below), try to move it and see if there is a correlation in between you moving it and the arm getting idle, at last restore it if ever needed.
[image]
let us know

ems7, 2018-02-06:
Hi Julien,

First of all, the cameras are working, the firewire issue seems to be unrelated to the collapsing.

After some testing, I can confirm that the robot collapses when only the yarpserver, and robot interface are running, so the problem isn't caused by any software further down stream than these (i.e. other yarp modules).

Upon starting the robot interface, it can be seen that the left shoulder and elbow encoders are not working properly due to an issue with the AEA magnetic position sensors. 

Opening the motor gui, the left shoulder and elbow motor control boards are in hardware fault status, they can be idled and run but shortly return to hardware fault status, whether the arm is moved or not. I moved the arm through its full range of motion of all four joints (should roll, pitch, yaw and elbow), the position of the arm did not seem to affect the control boards going into hardware fault status.

Further to this, sometimes randomly, the same issue with the AEA magnetic sensors on the torso joints occurs and the robot collapses forward as the torso motors stop working.

here are the logs: [https://cloud.ingo-keller.de/index.php/s/yiioevY50xYHTAE](url) 

Looking in the log it seems that there is a message problem on the CAN bus prior to the collapsing / AEA magnetic sensor issue. It is not clear if the trouble on the CAN bus causes wrong sensor readings or if a broken sensor corrupts the data on the CAN bus. 

Important lines from the log:

> 918 8,985409 <UNDEFINED> WARNING�[00m] from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 239s 767m 970u: (code 0x0000000b, par16 0x01fb par64 0x0021012100920080) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
>  ...
>  1162 15,873289 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 246s 650m 113u: (code 0x0000000d, par16 0x01b0 par64 0x0009002200b100d9) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
>  ...
>  2614 526,589534 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 757s 358m 499u: (code 0x02000008, par16 0x0000 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2615 526,589658 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 757s 358m 638u: (code 0x02000008, par16 0x0002 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2616 526,589704 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 757s 359m 500u: (code 0x02000008, par16 0x0001 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2617 526,589761 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 757s 359m 664u: (code 0x02000008, par16 0x0003 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2618 526,589795 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 757s 360m 104u: (code 0x0000000d, par16 0x01a6 par64 0x00050122004301c3) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
>  ...
>  2623 528,501810 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 759s 273m 377u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2624 528,501934 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 759s 273m 513u: (code 0x0200000a, par16 0x0001 par64 0x0000000000000003) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2625 528,501979 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 759s 273m 655u: (code 0x0200000a, par16 0x0002 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
>  2626 528,502015 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 759s 273m 792u: (code 0x0200000a, par16 0x0003 par64 0x0000000000000007) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
>  ...
>  2643 534,425779 <WARNING>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src CAN2, adr 13, time 765s 202m 29u: (code 0x0000001b, par16 0x62d0 par64 0x00000000010001

Any help with solving this issue would be much appreciated,
cheers,
Eli

julijenv, 2018-02-08:
Hi @ems7 ,
here is the all list of error and warnings as well as time in between the different event and from that I will indicate you a procedure to try to go deeper in the debug

> 232s 125m 450u      mc initialised
239s 766m 43u       strain initialised
246s 650m 113u      warning tx phase is too long
Left_Arm_Calibrator is finished

Until there everything is perfectly alright
But after about 9 minutes:
> 757s 358m 499u      AEA errors on all 4 joints (MC: AEA encoder invalid data)
757s 358m 638u
757s 359m 500u
757s 359m 664u
757s 360m 104u      warning of tx phase longer than allowed: it is 422 microsec (max is 300), but is due to previous 4 prints, maybe ...
spikes now
759s 273m 377u      MC: AEA encoder has spikes for the 4 joints of the left arm
759s 273m 513u
...
759s 273m 792u
invalid data for the 4 joints of the left arm
762s 358m 517u
762s 358m 653u
762s 359m 514u
762s 359m 661u
can error at
765s 202m 29u       SYS: EOtheCANservice could not parse a rx can frame (frame is: par16 0x62d0 par64 0x0000000001000100), comes from can2 (strain)      
MC: AEA encoder invalid data
767s 359m 514u
...
...
...
MC: AEA encoder invalid data
772s 360m 514u
...
...
...
MC: AEA encoder invalid data
777s 361m 517u
etc. every 5 seconds
824s 448m 29u       SYS: EOtheCANservice could not parse a rx can frame from CAN2:13 (par16 0x62d0 par64 0x0000000001000100)
spikes
866s 273m 377u      MC: AEA encoder has spikes for 4 joints
867s 379m 514u      invalid data for AEA encoders for the 4 joints of the left arm

So from all of this I would like you to test 2 things.
First, relaunch the yarprobotinterface , as you probably did it (the first time - log you sent us for this thread issue). When the left arm collapse, please relaunch it by idle and run for 3, 4 times every time is gonna collapse to get a maybe time stamp of the problem. close the robotinterface, switch off the motor wait 15-20 minutes that the motors cool down.
And redo First. Sedn us the both log file.

Secondly, please do a 'git log' for the repositories
icub-main, yarp and icub-firmware-build on pc104
thx

ems7, 2018-02-08:
Hi @julijenv,

here is a google drive with all the logs. We managed to get the arm to repeatedly collapase and the torso collapsed twice, once on each trial.

Further to that there is an issue with the right arm (hand) motors which are becoming idle during the calibration process and when restarted using the motor gui, their velocity is set to zero, I set the velocity to 60% to be inline with the other parts of the arm but when homing the right arm, the velocity jumps to 100% for these parts (please see the images rightarmmotorgui_8_2_18 and rightarm2motorgui_8_2_18 in the google drive).

The logs for the robot interface are called  log_pc104_yarprobotinterface_2485_8_2_18 and log_pc104_yarprobotinterface_1532_8_2_18b for the first and second tests respectively.

Thanks again!
Eli

julijenv, 2018-02-08:
Hi @ems7 ,
please consider using and install Teamviewer on the laptop in orde rto be able to have a better look on the robot and the different issues, I am not so much worry about the right arm yet or the otrso may main preoccupation is the left arm. Let me know when available to investigate, unfortunately not bwfore monday I fear, I have a deadline on bureaucracy stuff tomorrow.
PS have a skype account working also and try to get in contact with our skype ID : icub.support

ems7, 2018-02-15:
Hi @julijenv,

I won't be able to use Teamviewer as it doesn't work with the university firewall.

Could we organise a skype session for next week on Wednesday 21/2/18 to have a look at the robot?

thanks,
Eli

julijenv, 2018-02-15:
Hi @ems7 ,
I was wondering when you will be getting in touch with me. Please ring me the 21 because I should have problably already forgotten and also give me a time spam that day on which it is more probably to skype.
c u soon

ems7, 2018-02-15:
Hi @julijenv,

I am available all day next wednesday, so we can skype at whatever time suits you.

best wishes,
Eli

ems7, 2018-02-19:
@julijenv,

Will we Skype this Wednesday then? I can do 9am GMT (10am Italian time).

I am also available later in the day if you prefer. 

Best wishes,
Eli

julijenv, 2018-02-20:
@ems7,
seems perfect to me as well, C U 2moro

ems7, 2018-02-21:
@julijenv 

Are you available now?

julijenv, 2018-02-21:
debugging sesision N1 with @ems7,
here is the following logfile requested by me (@julijenv) 
[log_pc104_yarprobotinterface_1485_no_torso.txt](https://github.com/robotology/icub-support/files/1744936/log_pc104_yarprobotinterface_1485_no_torso.txt)
[log_pc104_yarprobotinterface_1523_no_right_arm.txt](https://github.com/robotology/icub-support/files/1744937/log_pc104_yarprobotinterface_1523_no_right_arm.txt)
[log_pc104_yarprobotinterface_1666_no_torso.txt](https://github.com/robotology/icub-support/files/1744938/log_pc104_yarprobotinterface_1666_no_torso.txt)
[log_pc104_yarprobotinterface_1934_no_torso_no_left_arm.txt](https://github.com/robotology/icub-support/files/1744939/log_pc104_yarprobotinterface_1934_no_torso_no_left_arm.txt)
[log_pc104_yarprobotinterface_2076_no_torso_no_left_arm.txt](https://github.com/robotology/icub-support/files/1744940/log_pc104_yarprobotinterface_2076_no_torso_no_left_arm.txt)
[log_pc104_yarprobotinterface_2181_no_torso_no_left_arm.txt](https://github.com/robotology/icub-support/files/1744941/log_pc104_yarprobotinterface_2181_no_torso_no_left_arm.txt)
[log_pc104_yarprobotinterface_2367_no_right_arm.txt](https://github.com/robotology/icub-support/files/1744942/log_pc104_yarprobotinterface_2367_no_right_arm.txt)
[log_pc104_yarprobotinterface_2579_no_right_arm.txt](https://github.com/robotology/icub-support/files/1744943/log_pc104_yarprobotinterface_2579_no_right_arm.txt)
[log_pc104_yarprobotinterface_2740_no_right_arm.txt](https://github.com/robotology/icub-support/files/1744944/log_pc104_yarprobotinterface_2740_no_right_arm.txt)

here some results and findings:

- log_file_1485_no_torso

 -> robot didnt start coz pblm with an extra calling for this file 

>  ERROR  Target device **torso-mc_wrapper** (network = torso ) does not exist.

- log_file_1523_no_R_A

 -> pblm with torso joint 2 = AEA2 =>> physical joint0 on yarprobotmotorgui (yaw)
> ERROR  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 294s 667m 163u: (code 0x02000008, par16 **0x0002** par64 0x0000000000000002) -> MC: AEA encoder invalid data. 
> <WARNING>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 291s 179m 73u: (code 0x0200000a, par16 **0x0002** par64 0x0000000000000001) -> MC: AEA encoder has spikes.

[image]
Please check the connection form sensor labelled 4S0 at the bottom part of the robot, the cable gets out from the ombilical cord and reach the board EMS5 on the bottom part of the robot, it is also section  as seen on the wiring (see picture above)
to be continued

ems7, 2018-02-22:
Logs from right arm testing: 

https://drive.google.com/drive/folders/1fbf68rL7N8BF3sc18B700NHdRrTHELsE

Running the same test three times, the robot interface failed two times after the right arm finished calibrating, with error:

>ERROR	positionMoveRaw: skipping command because BOARD right_arm-eb28-j8_11 IP 10.0.1.28  joint  3  is not in VOCAB_CM_POSITION mode

The first time, the robot interface crashed slightly later, though this error was still present.

>1268	63,296731	ERROR	positionMoveRaw: skipping command because BOARD right_arm-eb28-j8_11 IP 10.0.1.28  joint  3  is not in VOCAB_CM_POSITION mode
1295	63,685302	ERROR	A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD right_arm-eb28-j8_11 IP 10.0.1.28, joint 3, current mode: cal, requested: idl
1296	63,685431	ERROR	failure of embObjMotionControl::checkRemoteControlModeStatus(j=3, targetmode=idl) for BOARD right_arm-eb28-j8_11 IP 10.0.1.28 after 17 attempts and 0.261167 seconds
1297	63,685475	ERROR	In embObjMotionControl::setControlModeRaw(j=3, mode=idl) for BOARD right_arm-eb28-j8_11 IP 10.0.1.28 has failed checkRemoteControlModeStatus()

julijenv, 2018-02-22:
Hi @ems7 ,
[log_pc104_yarprobotinterface_1565_right_arm_1.txt](https://github.com/robotology/icub-support/files/1747658/log_pc104_yarprobotinterface_1565_right_arm_1.txt)
[log_pc104_yarprobotinterface_1705_right_arm_2.txt](https://github.com/robotology/icub-support/files/1747659/log_pc104_yarprobotinterface_1705_right_arm_2.txt)
[log_pc104_yarprobotinterface_1854_right_arm_3.txt](https://github.com/robotology/icub-support/files/1747660/log_pc104_yarprobotinterface_1854_right_arm_3.txt)

- for the first log_1565,
->actually it crashes because of the torso
> 1196 63,000517 **ERROR** Target device torso-mc_wrapper (network = torso ) does not exist. 

I think if we want to perfectly debug the rihgt _Arm we need to get rid of all the other joint, please remove all the torso parts.

- for log_1705
->same as previously, and the yarprobotinterface also wehn closing, a lost of memory, big crash

-for log_1854
->same

ems7, 2018-02-22:
Okay, new round of testing:
logs are here: https://drive.google.com/drive/folders/1SklQOCUCk7TCmNCcSZrdik0D15JSPLck
I have added 
<build="1"
 to line 4 of icub_all_no_torso_no_left_arm.xml
this has gotten rid of multiple warnings about syntax errors in other xml files as this parameter was assumed to be zero as it was missing and other files have it set to 1. (Could a software person explain what this parameter does and perhaps why it was missing from  this xml file? Should it not be automatically generated when the file was created? It appears in all other xml files for example  /home/icub/.local/share/yarp/robots/iCubEdinburgh01/cartesian/right_arm-cartesian.xml

I have commented out
> <elem name="torso">torso-mc_wrapper</elem
from /home/icub/.local/share/yarp/robots/iCubEdinburgh01/cartesian/right_arm-cartesian.xml

this has gotten rid of the error from the robot interface trying to load the torso mc wrapper

The robot interface still crashes, and there are now issues with both legs for example:
> 1470 139,647924 <ERROR> Left_Leg_Calibrator : joint  4  is idle, skipping park [[01;32mDEBUG[00m]head-calibrator starting park device head-mc_wrapper
see file log_pc104_yarprobotinterface_1557_right_arm_build_1_no_torso_wrapper_in_right_arm_cartesian_xml

I then removed the legs from the icub_all...xml 
the robot interface still crashes:
>2246	542,310851	ERROR	In embObjMotionControl::setControlModeRaw(j=3, mode=idl) for BOARD right_arm-eb28-j8_11 IP 10.0.1.28 has failed checkRemoteControlModeStatus()

see file log_pc104_yarprobotinterface_1780_right_arm_build_1_removed_legs_no_torso_wrapper_in_right_arm_cartesian

I then added the torso back into both icub_all...xml and right_arm-cartesian.xml
There is an error with torso-eb5-j0_2, the torso calibrator and the right arm calibrator.
>3218	2035,023116	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 2156s 43m 577u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .

see file log_pc104_yarprobotinterface_1920_right_arm_torso
The interface did not crash

rerunning with the same configuration after letting the motors cool down for 10 minutes:
same issues as before
see file log_pc104_yarprobotinterface_2109_right_arm_torso_cold_start

ems7, 2018-02-22:
I have also found a lose connection for the cable 4S0 as you suggested I would. I have reattached the loose wire, but I think it will need dismantling and recrimping in the future.

julijenv, 2018-02-22:
Hi @ems7 ,

log_file_2109
-> robot has EXTERNAL FAULT ACTIVATED on the torso
> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 98s 673m 444u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .

-> I would suggest you to take a look at the bottom of icub and look for the big white wire which and find the connector flying around and make sure that all the connection are alright onto it.

log_file_1557
-> EXTERNAL FAULT for legs
> **ERROR**  from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 133s 200m 237u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
**ERROR**  from BOARD 10.0.1.7 (left_leg-eb7-j4_5), src LOCAL, adr 0, time 148s 360m 222u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
**ERROR**  from BOARD 10.0.1.8 (right_leg-eb8-j0_3), src LOCAL, adr 0, time 148s 640m 618u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
**ERROR**  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 148s 794m 771u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .

As for before suggested have a look at the white wire, I guess it has been pinched or snapped at the level of the connector Hopefully.
But the real problem is this
> 1444 139,637762 **INFO** ***** Attaching drivers to cartesian controller icub/cartesianController/right_arm *****
1445 139,637795 **INFO** Received list of 1 driver(s)
 1446 139,637825 **ERROR** Expected list of 2 driver(s)
 1447 139,637854 **ERROR** Device right_arm_cartesian cannot execute attach 
 1448 139,637883 **ERROR** Cannot run attach action on device right_arm_cartesian

-> problem inside the cartesian controller config file

log_file_1780
-> problem with the encoder value for joint 11
> **DEBUG** Right_Arm_Calibrator : set 4 j 11 : Calibrating... enc values AFTER calib:  101.997 

it should be something close to 0 NOT 100 so the GUI disables axes set3  = joint 8,9, 11, 13

log_file_1920:
same as previously mentionned

ems7, 2018-02-23:
Logs and other files are here: https://drive.google.com/drive/folders/1fbf68rL7N8BF3sc18B700NHdRrTHELsE  in folder right arm tests/friday

I found the disconnected white wire and reconnected it, the legs are now fine.

Checking the right arm, first with no legs, same issue as yesterday with joint 11.

Checking right arm with legs, legs are working joint 11 is still a problem.

Looked into right_arm-calibration.xml - only joints 0 - 9 are calibrated here.

Looking in hardware/right_arm-eb28-j8_11-mc.xml and hardware right_arm-j8_11-mc.xml  and
comparing  to the left hand equivalent of these files, all the numbers seem to match with the exception of a few switches of sign, presumably to account for the direction of the hands being different with respect to the robot coordinate system. I have uploaded these xmls to the google drive.

I have looked through the mechanical, electronics, FT, intertial and MAIS xmls with regards to joint 11 on the right arm and found nothing that doesn't match the left arm

julijenv, 2018-03-21:
Hi Eli,
It has been performed by Barry the resoldering of the sensor AEA of the elbow.
It has been discovered that the rigt index proximal magnet was off due to untightened screw, after having performed repositionned of the magnet and tightned the screw , the yarprobotinterface wasn't given any errors when launching the robot with left arm disactivated.
BUT after having reput back the cover of the rightupper arm it appeared that ine cable got stretched and a new error appeared.
here is the findings of this issue:
the cable stretched is coming form this sensor
[image]
to reach it, just remove the six screws holding the yello wpiece to the red piece. 
[image]

before performing the unmounting of the arm just make sure to disconnect the sensor and someone near to hold the arm and of course make sure to mark or sign the position of the red and yellow piece respect to each other.
let me know

ems7, 2018-03-26:
Hi Julien, after taking the right arm off with Barry, we can see that the cable we thought was broken, isn't broken. It seems that the insulation ends prematurely and that there isn't heat shrink wrap covering the end of the wire. Please see pictures.
Barry performed a continuity test all of the wires running through that cable to the sensor and they all worked.
[image]

[image]

Barry is happy to replace the cable and add more heat shrink wrapping but this seems like it is unlikely to fix anything. So we likely have another broken cable somewhere else on the arm.
I'm not sure how you would like to proceed, but perhaps we should discuss further you visiting again.
Cheers, 
Eli

ems7, 2018-03-26:
More Images: 
[image]
[image]
[image]

DanielePucci, 2018-03-27:
CC
@gabrielenava @maggia80

julijenv, 2018-03-28:
Hi @ems7 ,
from what I can see  I guess the first log has been launched a bit too fast, too much error due to retarting board, so I guess you may consider relaunching the the robot less fast that ususal to prevent from having those basic error.
Next I can see a lot of error due to CAN1 and CAN 2 for the board 27 in the upper arm, please check the connection.
let me know

ems7, 2018-03-28:
Hi Julien, I've added some more logs now that the broken wire has been recrimped.

The arm still runs fine without the cover, with the cover attached and plugged in there are errors with eb27 J_4 but also with joint 3 now which is in hardware fault status.

julijenv, 2018-03-29:
Hi @ems7 ,
file 1839_noRA_noLcover
NO ERROR

file 2247_noLA_rightcover_unpl : error on head j0 of face
> **ERROR** from BOARD 10.0.1.22 (face-eb22-j0_1), src LOCAL, adr 0, time 206s 318m 433u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .

file 2288_noLA
> **ERROR**  from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 837s 662m 720u: (code 0x02000001, par16 0x0001 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
- I think that the robot start too fast there are too many 
> from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 837s 454m 899u: (code 0x00000034, par16 0x0001 par64 0x0000000000000039) -> SYS: **the board has detected a restart of the remote transceiver** because the rx ropframe sequence number is 1. In par64 there is the expected number. + .

file 2452-noLA_no_cover
NO ERROR

file 2499_noLA_no_cover
NO ERROR

file 2504_nLA_no_cover
> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 307u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .

> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 432u: (code 0x02000001, par16 0x0001 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .

> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 545u: (code 0x02000005, par16 0x0001 par64 0x0000000000002c00) -> MC: 2FOC CAN generic error. Errors happened in the CAN bus between the EMS and the 2FOC board. par16 = ID of joint. + . 

> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 684u: (code 0x0200000b, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder dirty. The number of thicks in a complete revolution of the motor was lower than expected, the optical disks need to be cleaned. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .

>  **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 827u: (code 0x0200000c, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder index broken. The reference special thick was not detected during a complete revolution of the motor, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + . 

> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 327m 959u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .

> **ERROR**  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 25s 328m 54u: (code 0x0200000e, par16 0x0001 par64 0x00000000ffe9c222) -> MC: generic motor error (see 64 bit mask parameter). + .
- Could you check if the problem we had with the white cable is still fixed? Strangely enough it is the first error message that appears. and afterwards the boards just make a lot a complaints that could be due to the first one.

file 2709_noLA_cover plugged
> **ERROR**  from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 323s 771m 772u: (code 0x02000002, par16 0x0003 par64 0x0000000000000000) -> MC: i2t limit exceeded. The motor has been turned off to prevent it from being damaged by overheating due to a continuous high current. par16 = ID of joint. + 
- Could you check the motor 2M3 wiring?

Let me know, wiating for u behind skype

julijenv, 2019-10-01:
Hi @ems7,
any news, sorry this issue is openend since wuite a long time, but I wtill didnt get any reply form you.
Let me know

closed 2019-10-02 by maggia80