robotology/mesh-support #595: Need to change Icub IPs

jypuigbo, 2018-01-29:
Hello, 
As you might know, icubBarcelona has moved to a new institution. The new IT services are telling us that they can't give us the IP range 10.0.0.XX. Is there the possibility to change the IPs to the range 10.4.131.XX?

If Positive: Is there any important thing we need to consider that might not be that obvious to the uninitiated in the networks' world? especially regarding YARP or the PC104?

If Negative: We'll keep fighting them to get our 10.0.0.XX network. 

Thanks a lot, 

Jordi & Alex

mbrunettini, 2018-01-31:
It is possible to change the IP configuration to a different subnet, but you have to change also all the configurations on all iCub enviroment (PC104, laptops, servers - if present - and other machines) to reflect the new IP range

gsaponaro, 2018-01-31:
Hi @jypuigbo , we do the same type of network customization in IST Lisbon. Here is how we usually proceed, both on the pc104 and on the other machines connected to the robot.

**/etc/network/interfaces**
Set up the desired address, netmask, gateway, (optionally) DNS, etc. For example in our case we end up with this kind of file. In your case you have to put 10.4.131.x values.
```
auto lo
iface lo inet loopback
auto eth0
iface eth0 inet static
address 10.10.1.100
netmask 255.255.255.0
network 10.10.1.0
broadcast 10.10.1.255
gateway 10.10.1.254
dns-nameservers 10.0.0.1 10.0.0.2
```

**/etc/hosts**
Comment out (or remove) the lines referring to the setup which is forbidden at your institute, add your desired lines instead. For example:
```
# IIT
#127.0.1.1	pc104.icub.local pc104
#10.0.0.1 	icub-srv icubsrv
# IST
127.0.0.1       localhost pc104
127.0.1.1	pc104.visnet pc104
10.10.1.50	pc104
10.10.1.51	icub-cuda
```
Almost done. Now a couple of further changes specific to the pc104 are necessary. (Note: the filenames below are valid for the iCub LIVE image 6.0, they might be slightly different in the [latest version 7.9](https://github.com/robotology/QA/issues/268)).

**/etc/rciCub.d/S90_mount-remote-fs.sh**
Change the following line
`PING_DEST="10.0.0.1"`
to point to the IP address of your NFS server (i.e., of the machine exporting the NFS shares in the cluster, if available)

**/etc/rciCub.d/mounts.list**
Similarly, replace 10.0.0.1 with your custom NFS server IP.

Related issues: https://github.com/robotology/QA/issues/136, https://github.com/robotology/icub-support/issues/494

Also, quoting @mbrunettini from [this comment](https://github.com/robotology/icub-support/issues/494#issuecomment-349634263):

> **A note** : usually the iCub is used with a laptop (or a server) that acts as gateway and segregate the iCub subnet from the rest of the world; this helps network bandwith performances.

mbrunettini, 2018-04-10:
can we close this?

julijenv, 2018-07-19:
can we close this issue?  @jypuigbo