robotology/mesh-support #597: Robot can not start with Xsens IMU

fiorisi, 2018-02-01:
The robot can not start. The error message is "xsens: can not open serial port".

Temporary we commented the [xsens sensor declaration](https://github.com/robotology/robots-configuration/commit/917f06f87bfd0ece74731fb7fef0c4bb7ad0e137#diff-3e178dfcf159536f87b37af52a2e16d5R44) to be able to turn the robot on.

traversaro, 2018-02-01:
Are you sure that this is not related to the udev rules? Unfortunately I forgot to propagate them to the icub images.

julijenv, 2018-02-06:
Hi @fiorisi,
era quello che suggereva @traversaro ?!?!?
se si please close the issue, thx in advance

nunoguedelha, 2018-02-06:
More information in this [issue comment](https://github.com/loc2/element_icub-at-ihmc/issues/39#issuecomment-355225289)

traversaro, 2018-02-06:
Before we commit it to a more permanent position, the udev file is the following: 
Location: 
~~~
/etc/udev/rules.d/99-usb-serial.rules
~~~
Content (see http://www.linux-usb.org/usb.ids):
~~~
SUBSYSTEM=="tty", ATTRS{idVendor}=="2639", SYMLINK+="ttyXsens"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", SYMLINK+="ttyFTDI"
~~~

nunoguedelha, 2018-02-06:
- I've run a quick test, starting successfully the robot interface with the Xsens IMU in the waist activated
- could then connect to the respective Yarp port and display the IMU measurements.

Closing the issue.

closed 2018-02-06 by nunoguedelha