robotology/mesh-support #599: Right hand fails to calibrate at startup

S-Dafarra, 2018-02-08:
When starting the robot the right thumb fails to calibrate and I get the following error
```
147896	7137,261900	ERROR	unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
148072	7169,020714	ERROR	Right_Arm_Calibrator : checkGoneToZeroThreshold: joint  8  Timeout while going to zero!
148079	7169,030884	ERROR	Right_Arm_Calibrator : set 4 : some axis got timeout while reaching zero position... disabling this set of axes
```
cc @julijenv @gabrielenava

S-Dafarra, 2018-02-20:
Here a more detailed log of a similar issue
[log_icub-head_yarprobotinterface_11690.txt](https://github.com/robotology/icub-support/files/1741833/log_icub-head_yarprobotinterface_11690.txt)

julijenv, 2018-03-07:
Hi @S-Dafarra ,
After having restored the right distal joint of the index, did this issue reoccur? if not please close this issue thx

S-Dafarra, 2018-03-07:
It is better now. Sometimes it has some problems on ``r_hand_fingers``, but we can close this issue.

closed 2018-03-07 by S-Dafarra