robotology/mesh-support #603: Robot goes in hardware fault when in torque control

S-Dafarra, 2018-02-21:
This morning the robot went in hardware fault four times when controlling it in torque. We saw several error like:
```
20643 7389,589567 <ERROR>  from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 1085s 394m 169u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
```
Here a full log of the last run
[log_icub-head_yarprobotinterface_4540.txt](https://github.com/robotology/icub-support/files/1743657/log_icub-head_yarprobotinterface_4540.txt)
@gabrielenava @DanielePucci @traversaro @marcoaccame @julijenv

S-Dafarra, 2018-02-21:
Actually both! But I don't know how it could be related :thinking:

S-Dafarra, 2018-02-21:
Ah no sorry. I haven't changed the IMU used inside wholebodydynamics.

julijenv, 2018-03-07:
Hi @S-Dafarra ,
I dont know if this issue is always on going on this robot, but by looking into the log file I fear that it isnt related to the FT sensor itself but more related to the communication (for example this message):
- **ERROR**  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 241s 643m 505u: (code 0x00000030, par16 0x0002 par64 0x0000000000004098) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
 this usually means that something on the ethernet line in between the boards is not going well and create a hole inside the packets and consequently everyboard detect it and start printing out those messages which start to congest the line and as the loop for the FT sensor is 100ms, if the piors messages with higher priority prevent the control loop to work , the robot goes in HW Fault.
@marcoaccame , could you try to tell me if I am right by saying that the problem is in between the EMS6 and EMS7.
thx in advance

S-Dafarra, 2019-10-03:
I think this issue may be considered fixed. We will open another one if it occurs again.

closed 2019-10-03 by S-Dafarra