robotology/mesh-support #605: Robot goes in hardware fault also in position control

S-Dafarra, 2018-02-26:
It happened a couple of times. After a while the robot just falls and the motor led starts blinking. Probably is due to some connection problem in the right arm since the first error message is the following:
```
 from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src LOCAL, adr 0, time 468s 134m 981u: (code 0x06000001, par16 0x0001 par64 0x3000000000000000) -> ETH monitor: link goes down in port specified by par16 (0 = P2, 1 = P3, 2 = internal).  Application state is most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb3-j0_3 IP 10.0.1.3 and nv ID32 = 0x0100010e -> IND = 1, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
550395	5076,166394	WARNING	EthResource::getRemoteValue() cannot have a reply from BOARD right_arm-eb3-j0_3 with IP 10.0.1.3 at attempt # 1 w/ timeout of 0.2 seconds
  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb3-j0_3 with IP 10.0.1.3  even after 0.200104 seconds: CANNOT PROCEED ANY FURTHER
```
Here also a full log:
[log_icub-head_yarprobotinterface_7306.txt](https://github.com/robotology/icub-support/files/1758865/log_icub-head_yarprobotinterface_7306.txt)

@marcoaccame @julijenv @gabrielenava @DanielePucci

marcoaccame, 2018-02-26:
Yes, @S-Dafarra: well spotted! It is the link in exit from board eb24. 

I have already talked to @julijenv  about that.

julijenv, 2018-03-07:
Hi @S-Dafarra ,
did the issue continue poping up, or did my magic work on it?

S-Dafarra, 2018-03-07:
It actually appeared once again but the logger was already saturated.

S-Dafarra, 2018-03-15:
I am not sure, but probably this issue is related to https://github.com/robotology/icub-support/issues/614 . @marcoaccame @julijenv if you think they may be related, maybe we can close also this one.

julijenv, 2018-03-15:
Hi @S-Dafarra ,
the real question is : 
is it still a problem or is the problem stop appearing?

gabrielenava, 2018-03-15:
It seems to me that this happened again, but while the robot was torque controlled. Here is the logger:

[log_icub-head_yarprobotinterface_3585.txt](https://github.com/robotology/icub-support/files/1815464/log_icub-head_yarprobotinterface_3585.txt)