robotology/mesh-support #606: Loose hip pitch cables

S-Dafarra, 2018-02-26:
While walking, the cable on the right hip pitch became very loose. Talking with @julijenv, he noticed that one of the tensioner screws is stripped.
The left hip pitch instead has probably a too long cable. This can cause backlash and increase the probability for the cable to exit.

@DanielePucci

fjandrad, 2018-02-26:
> The left hip pitch instead has probably a too long cable.

For sure is too long it also complicates getting clean data for calibration of F/T sensors on the pole. Since it can get smashed between the leg and the cover of the hip in some positions.

S-Dafarra, 2018-02-26:
@fjandrad sorry I should have been more precise. I was meaning the mechanichal one, but I agree that also the electrical cables are quite long.

julijenv, 2018-05-09:
to be lcosed due to same issue as 647

closed 2018-05-09 by julijenv