robotology/mesh-support #607: Problem in the right leg

diegoferigo, 2018-02-27:
After some experiments controlling the right ankle in torque mode, the `yarprobotinterface` started failing with the following error:

```
612	2,747582	ERROR	 from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 97s 930m 373u: (code 0x05000016, par16 0x2003 par64 0x0000000000ff0033) -> CFG: EOtheEncodeerReader cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
622	2,838549	ERROR	 from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src CAN2, adr 0, time 98s 22m 107u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
624	2,838567	ERROR	 from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src CAN2, adr 0, time 98s 22m 607u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
625	2,848605	ERROR	 from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src CAN2, adr 0, time 98s 25m 106u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

```

julijenv, 2018-02-27:
Hi @diegoferigo ,
could you please take a look at the roobt to know if it wors now?
I had to resolder one wire so I guess that should be the problem uncountered.
let me know and close ths issue

diegoferigo, 2018-02-27:
Hi @julijenv,

Everything looks perfect, thanks as usual! @gabrielenava will update this issue if something else pops out from his experiments.

closed 2018-02-27 by diegoferigo