robotology/mesh-support #608: Right shoulder cable broken

gabrielenava, 2018-02-27:
During Yoga experiments with `iCubGenova02`, the right shoulder transmission cable broke.

julijenv, 2018-02-28:
Hi @gabrielenava ,
Please check if the reobot is now working properly and if yes please close the issue.
The cable N7 has been fixed as well as a total retensioning of the shoulder cables from 1 to 8 on both the  shoulders.
It has been taken from warehouse:
IITCODE Qta Description
4764 x 1 : shoulder rotation cable N7

julijenv, 2018-02-28:
Hi @gabrielenava ,
Please check if the reobot is now working properly and if yes please close the issue.
The cable N5 has been changed coz was too much pinched
It has been taken from warehouse:
IITCODE Qta Description
4762 x 1 : shoulder elevation cable N5

gabrielenava, 2018-03-07:
the robot is now working properly. Closing the issue.

closed 2018-03-07 by gabrielenava