robotology/mesh-support #609: iCubErzelli01: Neck is stuck looking a 3D point using iKinGazeCtrl

ddetommaso, 2018-02-28:
### Description of the failure

I am using the iKinGazeCtrl to control the iCub gaze for pointing specific locations in the Cartesian space. Specifically I am interested on the looking at three positions defined as follows:

(-1.0, 0.0, 0.7): looking at the user
(-0.5, 0.4, 0.2): looking at the right object on the desk
(-0.5, -0.4, 0.2): looking at the left object on the desk

Using the iKinGazeCtrl default parameters the neck stuck whenever it switches between the "looking a the user" and "looking at the objects" (left or right). After this condition I need to restart the yarprobotinterface because it's not possible to control the neck properly anymore. Increasing the value of the neck trajectory time it works properly, but quite slow (i.e. it works for values > 1.4s).

Is it possible that the distance between the two 3D points is far enough to require a very high velocity value that exceeds the limits, so the neck is stuck to prevent damage ? If this assumption is correct, is there any way to use a dynamic trajectory time of the neck during the gaze tracking based on the distance between the current and the target points?

Thank you for the support

### Detailed conditions and logs of the failure

To test the issue I am running RPC commands to the port /iKinGazeCtrl/rpc

In the following you can see the joint positions and velocities in the condition looking from user location to the right object.

[image]

[image]

dump of the /icub/head/stateExt:o

[data.log](https://github.com/robotology/icub-support/files/1767982/data.log)

and logs

[log_pc104_yarprobotinterface_1552.txt](https://github.com/robotology/icub-support/files/1767990/log_pc104_yarprobotinterface_1552.txt)
[log_icubs4hri_iKinGazeCtrl_2048.txt](https://github.com/robotology/icub-support/files/1767991/log_icubs4hri_iKinGazeCtrl_2048.txt)

julijenv, 2018-03-07:
Hi @ddetommaso ,
Did u test the position you are targeting on real motion drive by yarpmotorgui? what I'd like to know is if the position can be reached by yarpmotorgui before requesting it through iKingazectrl (to uderstand if a mechanical/electronical problem or a software issue).
let us know

ddetommaso, 2018-03-08:
Hi @julijenv ,

both the target positions (left and right) are reachable from the starting position (front) because if I reduce the trajectory time of the neck everything is fine. For changing the trajectory time of the neck I mean that I change the default value of the iKinGazeCtrl parameter "trajectory_time::neck" (usually for values > 1.4s it works).

julijenv, 2018-03-08:
Hi @ddetommaso ,

Could you confirm me that whatever the speed you choose with the slider (0-100) on yarpmotorgui, you are able to move the head in the position you desired to reach. if yes, I doubt that the problem be mechanical or electrical but mainly software related.
let us know

ddetommaso, 2018-03-08:
Hi @julijenv,

I cannot confirm that because the same issue occurs without the iKinGazeCtrl. In the following you can see that after a while the neck stops (I was trying to move the pitch at the same speed of 70).

[image]

julijenv, 2018-03-08:
Hi @ddetommaso,
It looks like the neck got blocked, can u tell me if it does this over time (when motor get warm) or from a brand new start?
could you check the tension of the transmission belt on both side of the neck to see if they got worn or get loose? or misplaced?
let me know, I may have to pass by Erzelli to settle some problem on the red iCub I could on the same day passing have an eye on your robot.

ddetommaso, 2018-03-09:
Hi @julijenv ,

the issue occurs even from a brand new start. About the tension of the transmission belt both seem fine to me, but there is huge difference of friction.
Please let me know when you can come in Erzelli to have a look.

Thank you

julijenv, 2018-03-09:
Hi @ddetommaso ,
I should pass by on monday
I'll let u know if it gets postponed

ddetommaso, 2018-03-13:
Hi @julijenv ,

as we noticed the issue is not related to the iKinGazeCtrl because it's caused at a lower layer. Using the yarpmotorgui to move the neck is enough to get the issue. As an additional test I refer to the look 3D positions in the first message of this issue.

Unfortunately this issue still occurs very often and it's blocking our ongoing experiments. Please let me know if we have a chance to solve it in the next days or at least if we can minimize it's recurrence.

Thanks

julijenv, 2018-03-19:
Hi @ddetommaso ,
 I heard from Giorgio, that the things got worse, could you pleae try to relaunch the robot and send me a log file after trying to move the neck as you do normally until it blocks, thx in advance, some ethernet cable could be also loose. let me know

ddetommaso, 2018-03-19:
Hi @julijenv ,

please attached you can find the log.
Thanks

[log_pc104_yarprobotinterface_2078.txt](https://github.com/robotology/icub-support/files/1824457/log_pc104_yarprobotinterface_2078.txt)

julijenv, 2018-03-19:
Hi @ddetommaso ,
from those this picture of the internet board and connection I'd like you to ask EDL if they can this afternoon try to recrimp or check the cable labelled E20 in between the board EB20 and EB21 of the head.
[image]
[image]

It could be that the connector or the wire or a bit loose or need to be recrimped, let me know

ddetommaso, 2018-03-19:
Hi @julijenv ,

unfortunately at the moment we're running some test on the robot.
We can try to recrimp the cable tomorrow with you guys.

julijenv, 2018-03-20:
looks like it has bben restored,
one ball bearing of the left side of the neck was messy
IITCODE Qta DEscription
2695x1 : ball bearing

It has been also performed a recrimpping of the cable E20 fro the ground cable on both side.

closed 2018-03-20 by julijenv