robotology/mesh-support #610: Right Hand Fingers broken + Skin Right arms

reafrancesco, 2018-03-01:
### Description of the failure
- The right thumb of the hand is broken, a cable is loose provoking a noisy sound and endless motion.
- The right forearm skin is not working properly, dead taxels.

### Detailed conditions and logs of the failure
After motion with the hand (Pointing) for the thumb.
For the skin unknown.

julijenv, 2018-03-07:
Hi @reafrancesco , Jonas (you need to create an account github someday),
COuld it be possible to get a picture of the non responsive part of the skin? is it just some triangles or the entire skin patch?

pattacini, 2018-03-07:
@julijenv here's Jonas's username: @gonzalezJohnas

gonzalezJohnas, 2018-03-07:
Sorry @julijenv  was logg d with Francesco's account . Yeah I will provide the picture as soon as I can. It's the entire patch forearm and palm that is not responding. Might be cables disconnected

gonzalezJohnas, 2018-03-07:
Sorry for the last post was on my phone... Here are the list of fixes that we need for the iCub : 

- Skin Right Arm : The entire patches of the palm and forearm are not working

- Right hand : The oppose joint of the thumb is not working due to broken cable + Ring and pinky finger not closing properly.

- Skin Left Arm : Some taxels of the left forearm and upper arm are not responding

- Head : Some cables needs to be tight because there are loose  
  
@julijenv Tell me if you need more information

julijenv, 2018-03-07:
Hi @gonzalezJohnas ,
any picture for the head cables?!?!
taxels? you mean triangles? if yes just consider that nothing can be done except changed the entire patch (this is not something I can do and actually require to unmount the entire plastic cover to remove the faulty patch and reglued (done by external company)).
thx for your reply

julijenv, 2018-03-09:
Hi @gonzalezJohnas , @reafrancesco 
I should pass by on moday.
I'll let u know if it gets postponed.
cheers

julijenv, 2018-03-14:
it has been repaired the right hand , thumb abduction as well as the prozimal joint of the thumb it has been used 
IITCODE Qta DEscription
1815 x 2 m : carlstahl 0,63mm coated

it has been recrimped the skin connector.

closed 2018-03-14 by julijenv