robotology/mesh-support #611: setControlModes return false, even if the control mode switch appear to be successful

traversaro, 2018-03-09:
### Description of the failure
Under some condition that we are not fully able to debug, a `setControlModes (three parameters variant)` call for switching the control mode from position to torque is returning false, even if the control mode actually changes.

The failure only happens using a specific single joint controller based on the [wb-toolbox](https://github.com/robotology/WB-Toolbox), i.e. we were not able to reproduce it switching control modes using the `yarpmotorgui`. Even more interestingly, it does not affect all the joints, but just some joints, as specified in the following. 

| Joints affected by this problem | Joints immune to this problem |
|:-----------------------:|:-----------------------------------------------------:|
|  `l_knee`                |  `l_hip_pitch`  | 
| `l_elbow`               |  `l_ankle_pitch` |
| `l_ankle_roll`         | `l_wrist_prosup` | 

As this affected only the wb-toolbox based controller, we tried to add a delay before the https://github.com/robotology/wb-toolbox/blob/932c81cd049c7ce0c2494d60fe52bfd8c11dfd20/toolbox/src/SetReferences.cpp#L220 to see if this modified the situation, but it did not. 
However a successful workaround is to just ignore the return value, by commenting out the `return false` in https://github.com/robotology/wb-toolbox/blob/932c81cd049c7ce0c2494d60fe52bfd8c11dfd20/toolbox/src/SetReferences.cpp#L222 , meaning that apparently the switching is successful even if the return value is false.

### Detailed conditions and logs of the failure
### icub23 
yarp : 9c0833b7dac3faac1412eed6095b09a3a2df884d 26 Jan 2018 
icub-main: d613a19756e38710e2684987e5f5dde82f93cdb9 - 29 Jan 2018

### icub-head 
yarp: 8ae349c438096590b41ffa3d6a74a8f2ed548626
icub-head: 82cbac73253876010d3ecbcdf54309f694af22f1 
icub-firmware-build: ede90553cb872201ab550571c913d4454f290c96

This it the log of the yarprobotinterface :
~~~
17873	5284,987287		yarp: Receiving input from /WBT/SquareTorque2016blimitConfiguration/icub/left_leg/rpc:o to /icub/left_leg/rpc:i using tcp
17874	5284,997386		yarp: Receiving input from /WBT/SquareTorque2016blimitConfiguration/icub/left_leg/command:o to /icub/left_leg/command:i using udp
17875	5284,997409		yarp: Sending output from /icub/left_leg/stateExt:o to /WBT/SquareTorque2016blimitConfiguration/icub/left_leg/stateExt:i using udp
17876	5285,341037	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17877	5285,351130	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17878	5285,371386	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 4 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17879	5285,381513	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 5 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17880	5285,401770	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 6 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17881	5285,411897	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 7 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17882	5285,432050	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 8 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17883	5285,442126	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 9 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17884	5285,462247	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 10 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17885	5285,472306	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 11 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17886	5285,492423	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 12 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17887	5285,502479	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 13 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17888	5285,522646	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 14 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17889	5285,522671	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
17890	5285,532719	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 15 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17891	5285,552856	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 16 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17892	5285,552894	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
17893	5285,562930	WARNING	embObjMotionControl::checkRemoteControlModeStatus() has done 17 attempts and will retry again after a 0.010000 sec delay. (BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0) -> current mode = pos, requested = torq
17894	5285,583124	ERROR	A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD left_leg-eb7-j4_5 IP 10.0.1.7, joint 0, current mode: pos, requested: torq
17895	5285,583152	ERROR	failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=torq) for BOARD left_leg-eb7-j4_5 IP 10.0.1.7 after 17 attempts and 0.259189 seconds
17896	5285,583164	ERROR	setControlModesRaw(const int n_joint, const int *joints, int *modes) could not check with checkRemoteControlModeStatus() for  BOARD left_leg-eb7-j4_5 (IP 10.0.1.7)   joint  1  mode  torq
~~~

traversaro, 2018-03-09:
cc @marcoaccame @barbalberto @randaz81 @vibhoraggarwal

valegagge, 2018-03-12:
Hi,
maybe I fixed this issue. Please, see the branch EOMCfixSetControlModes of icub-main. It is a branch of master.
I wrote "maybe" because now i can't verify it.  So @traversaro , could you check it? thank you
Otherwise I'll check it, but  not before of friday.

cheers

barbalberto, 2018-03-13:
For sure that's a bugfix to be merged. Good catch

traversaro, 2018-03-13:
Thanks @valegagge for the prompt fix. As soon as we are back on the robot with @vibhoraggarwal we will test the fix.

vibhoraggarwal, 2018-03-13:
I tried to test the robot by uncommenting `return false`, but probably the issue still stands.

@traversaro @valegagge

traversaro, 2018-03-13:
Just checked with @vibhoraggarwal , and he actually he did not tested the branch pushed by @valegagge .

vibhoraggarwal, 2018-03-13:
Thanks @valegagge . The problem is fixed. 

P.S: Earlier I did not update `icub-head`, because of which it failed.

valegagge, 2018-03-23:
@vibhoraggarwal thanks for the feedback.

I'll merge the branch asap.

valegagge, 2018-03-26:
The bug fix is both in master and in devel branch.
I close this issue.

closed 2018-03-26 by valegagge