robotology/mesh-support #613: Board head-eb20-j0_1 does not respond after startup

gabrielenava, 2018-03-09:
### Description of the failure
 
after starting up the robot, an error associated to the `BOARD eb20-j0_1` appears in the logger, saying that 

```
4325	1271,595913	ERROR	  FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb20-j0_1 with IP 10.0.1.20  even after 0.200111 seconds: CANNOT PROCEED ANY FURTHER
```

the corresponding joints do not respond in the `yarpmotorgui`. The full logger is attached to the issue. 
[yarprunlog_09_03_2018_14_23_03.log.txt](https://github.com/robotology/icub-support/files/1797018/yarprunlog_09_03_2018_14_23_03.log.txt)

julijenv, 2018-03-16:
Hi @gabrielenava ,
is it possibile this afternoon to have a summary of what really is not working fine, on your robot?
thx c u soon

gabrielenava, 2018-03-18:
This issue is currently not repeatable on the robot. According to what discussed with @julijenv, I'll close this issue and we'll open it again if necessary.

closed 2018-03-18 by gabrielenava

aikolina, 2018-04-05:
Sorry if you get any notification. I'm using this issue to test some GitHub functionality.

[Green iCub's Journal](https://github.com/loc2/lab-organisation/issues/51).