robotology/mesh-support #614: Greeny fails to startup

S-Dafarra, 2018-03-12:
The robot does not seem to be able to start correctly. It failed three times in a row despite I rebooted the motor boards every time. It seems to be related to communication issues.

Here the log of the last attempt:
[log_icub-head_yarprobotinterface_5869.txt](https://github.com/robotology/icub-support/files/1803303/log_icub-head_yarprobotinterface_5869.txt)

@julijenv @marcoaccame @DanielePucci

S-Dafarra, 2018-03-12:
This is the output of ``FirmwareUpdater``
[image]

The state of all the boards displayed is running.

S-Dafarra, 2018-03-12:
Taking a look to the robot with @spada1978 and @maggia80, it seems that the problem is in the cabling of the right arm. In particular @maggia80 noticed that the ETH connector seems in a bad shape (bent of 90deg) and probably needs a replacement.

marcoaccame, 2018-03-13:
Also the log file reports problems in the eth cable of  the right arm: crc errors, link up and down, packet loss.

julijenv, 2018-03-15:
Hi @S-Dafarra, I checked the overall wiring insidethe upperarm and the only cable that popped up br trying to pulling them out was a power cable on the board EB27, I guess that if the board can't bridge (not enough power) the ethernet got broken.
Is it working now?

S-Dafarra, 2018-03-15:
We haven't had any new issues yet. I think we can close this issue. I will open a new one in case of an other failure. Thanks!

closed 2018-03-15 by S-Dafarra