robotology/mesh-support #615: Shoulder pitch slippage

fiorisi, 2018-03-16:
### Description of the failure
As shown in the following figure the coupling between the shoulder pitch joint and the gear output shaft is loose.
We experienced slippage more than one time.

[image]

Having a look at the CAD model seems that the torque is transmitted by a friction joint. Probably one or more screws are not tightened enough, or we need to insert a pin to prevent slippage.

[image]

cc @DanielePucci @S-Dafarra @gabrielenava

julijenv, 2018-03-19:
Hi @fiorisi , @gabrielenava , @DanielePucci ,
the fact that this piece slip is a normal and known  fact, the question is when do u realize that it slips that much, I mean is this situation appeared (the slippery in between the mark done and the moment you've noticed the slippery), were you holding the robot with your hand or is the robot just got this by normal used without falling (on the ground for exapmle as I could see in a picture lately).
late me know, if the last case, It might be due to untightend wcres as u suggested.

S-Dafarra, 2018-03-19:
Hi @julijenv! Unfortunately it happened quite often that the robot caused the slippage on its own after some very large movements, without anyone touching that arm or undergoing impacts. In the picture you saw the robot had no problem instead :smile:

julijenv, 2018-03-20:
has been fixed by retensing the screws of the piece lock the main shaft of the motor 0

closed 2018-03-20 by julijenv