robotology/mesh-support #619: icub-head restarts after few minutes every time is switched on

gabrielenava, 2018-03-26:
As you can see in the attached GIF, after being turned on the `icub-head` reboots. No one was touching the robot or was connected to `icub-head` via `ssh`. This happens every time the `icub-head` is switched on. Unfortunately, once it happened while we were compiling `yarp`, and we suspect there might be issues in the `icub-head` file system.

[image]

julijenv, 2018-03-28:
Hi @MrAndrea , this is your task, please fill up what you've done.

MrAndrea, 2018-03-28:
There was a problem on a voltage regulator in the control board BCB (the one with the 2 push buttons), probably due to iCub falls on the floor.
I changed the board with a new one (iitcode is 6117.A) and I checked that the robot works correctly.

S-Dafarra, 2018-03-30:
Thanks @MrAndrea. Now it seems to work fine!

closed 2018-03-30 by S-Dafarra