robotology/mesh-support #624: Joint encoders measurements have spikes

gabrielenava, 2018-04-05:
We tried the robot with a `yarp` branch different from `devel`, inside @valegagge fork. We had some problems running the Yoga demo because we were having spikes in the measured torque. By further investigation we noticed that the joint encoders were having spikes. In particular, from time to time, the encoders reading of a group of joints was going to zero for some instants. I'm going to update this issue with a picture of the spikes soon. In the end  we reverted to `devel` and we did not have this issue anymore.

DanielePucci, 2018-04-17:
@valegagge 
I think the link to the issue open in yarp is missing

valegagge, 2018-04-24:
@DanielePucci thanks for the ping.
Done!

valegagge, 2018-05-21:
fixed by https://github.com/robotology/yarp/pull/1668

closed 2018-05-21 by valegagge