robotology/mesh-support #626: Firmware update for iCubSingapore01

barbalberto, 2018-04-10:
### Description of the failure
yarprobotinterface does not start. A firmware update is required.

### Detailed conditions and logs of the failure
yarp and iCub repositories have been updated. This triggered a firmware update.
Support is requested to perform the update.

@julijenv can you help @tasbolat1 in this process?

tasbolat1, 2018-04-10:
In

> Compile iCub,

 which flags should be ON? (We're iCubSingpapore01)
```
You need to enable (all versions of iCub):

 ENABLE_icubmod_dragonfly2
 ENABLE_icubmod_skinWrapper
 ENABLE_icubmod_xsensmtx

CAN robots:

 ENABLE_icubmod_sharedcan
 ENABLE_icubmod_canmotioncontrol
 ENABLE_icubmod_canBusAnalogSensor
 ENABLE_icubmod_canBusDoubleFTSensor
 ENABLE_icubmod_canBusSkin
 ENABLE_icubmod_canBusInertialMTB
 ENABLE_icubmod_canBusVirtualAnalogSensor
 ENABLE_icubmod_cfw2can
 ENABLE_icubmod_parametricCalibrator

ETH robots: enable also the following

 ENABLE_icubmod_embObjAnalogSensor
 ENABLE_icubmod_embObjMotionControl
 ENABLE_icubmod_embObjSkin
 ENABLE_icubmod_embObjVirtualAnalogSensor
 ENABLE_icubmod_parametricCalibratorEth

Cartesian controller:

 ENABLE_icubmod_cartesiancontrollerclient
 ENABLE_icubmod_cartesiancontrollerserver

Versions from iCub 1.1.1 enable also:

 ENABLE_icubmod_cfw2can
 ENABLE_icubmod_skinprototype

Previous versions (iCub 1.0 and 1.1.0):

 ENABLE_icubmod_pcan: iCub 1.0 and 1.1.0
 ENABLE_icubmod_ecan: iCub 1.1.0
```

When we update the icub-main, we set some flags in `ccmake` to ON, which were OFF for old version.
We are doubting about that those flags may cause the problem also.  How can we make sure which flags should be ON for our icub?

barbalberto, 2018-04-10:
ENABLE_icubmod_dragonfly2
 ENABLE_icubmod_skinWrapper
 ENABLE_icubmod_xsensmtx

ENABLE_icubmod_embObjAnalogSensor
 ENABLE_icubmod_embObjMotionControl
 ENABLE_icubmod_embObjSkin
 ENABLE_icubmod_embObjVirtualAnalogSensor
 ENABLE_icubmod_parametricCalibratorEth
 ENABLE_icubmod_cfw2can
 ENABLE_icubmod_skinprototype

 ENABLE_icubmod_cartesiancontrollerclient
 ENABLE_icubmod_cartesiancontrollerserver

tasbolat1, 2018-04-10:
we got an error:
`cfw2can: cannot find cfw2canapi, turn off device`

julijenv, 2018-04-10:
disable cfw2can you have an ethernet board

julijenv, 2018-04-10:
please contact me on skype: ID  = icub.support

barbalberto, 2018-04-10:
ops, sorry

julijenv, 2018-04-10:
Hi @tasbolat1 ,
here some pictures for the sensor that makes some problem for the right leg

[image]
[image]
[image]
[image]
[image]
[image]
the list of tool needed for daily care of icub is in the following link
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/tool/iCubMaintenanceTools.xlsx

tasbolat1, 2018-04-12:
Thanks guys, firmware update has finished successfully!

closed 2018-04-12 by tasbolat1