robotology/mesh-support #627: Little fingers of icub (Singapore01) don't work properly

tasbolat1, 2018-04-13:
Hello everyone,
I have a following issue with icub's fingers:
Little fingers (joint 15) don't move when move with `yarpmanager` (logs give no error). And, sometimes only little finger moves (not ring one). How to check the reason of such behavior?

julijenv, 2018-04-13:
Hi @tasbolat1 ,
The joint probably run in the wrong direction or one of the 2 tendons is broken./ Could you take a philips (flat) screwdriver and spin it always to the external direction ( this should close both fingers).

tasbolat1, 2018-04-13:
@julijenv thanks for quick answer.
Which screw do I need to spin?  (you can orient me relative to the index finger screw)

tasbolat1, 2018-04-16:
Hi @julijenv! Are you around?

julijenv, 2018-04-23:
Hi @tasbolat1 , could you close this issue if consider as settled? if not please contact me

tasbolat1, 2018-04-23:
Hi @julijenv,
Yes, it's been solved. Thank you a lot

closed 2018-04-23 by tasbolat1