robotology/mesh-support #628: Loosen screws in the head neck

lornat75, 2018-04-13:
### Description of the failure
The screws that fix the head at the level of the pitch (the very small ones) need to be tighten. We manage to tighten the left one, but the right other is difficult to reach, without a long screw.

lornat75, 2018-04-16:
@karinne-ics-tum

julijenv, 2018-04-16:
Hi @lornat75 ,
As already explained over the phone the best way to tighten the little screws are first to remove the 2 hexagonal print screws above them and use a long screw driver to pass from the top of the head down to reach the screws remove them ona after the other, add some thread locker and retighten them STRONGLY. when done, please rescrew also the hexagonal ones and you are done.
let me knwow

julijenv, 2018-07-19:
any news on this matter??? @karinne-ics-tum

closed 2019-05-14 by julijenv