robotology/mesh-support #630: black images from left camera when eyes look far to the left

lornat75, 2018-04-13:
### Description of the failure
The problem is determinstic, when the eyes move to the left limit, the left camera goes black. It resumes as soon as the eyes move back. Probably the flat cable need to be changed.

lornat75, 2018-04-16:
@karinne-ics-tum

lornat75, 2018-04-16:
she may also need instructions on how to bend it... it is done with extreme skill :)

julijenv, 2018-04-18:
Hi @karinne-ics-tum ,
Basically try to reproduce the bending that have the old one, and just try to make sure that whenever the eye move from left to right the flat cable don't get smashed, or fold wrongly or badly which could affect the integrity of the flat cable. Fix the flat cable on the side of the frame of the head at a height which prevent bending of the cable and finally putthe other extremity inside the connector and flatten the cable to be able to close the head cover.
let us know

julijenv, 2018-07-19:
any news on this matter??? @karinne-ics-tum

closed 2019-10-12 by pattacini