robotology/mesh-support #631: head pitch becomes unstable when face tracker is activated with iKinGazeControl

lornat75, 2018-04-13:
### Description of the failure
When the facetracker is started the head pitch starts to oscillate wildely. 

I did check the PID in position mode (with yarpmotorgui) and the controller seems stable even at high speed. Therefore the problem could be due to either:

- Mechanical play (#628) to trigger oscillations
- Problem in iKinGazeControl (gains too high?)
- facetracker 

The network seems fine (images and encoders where running at framerate). I could not investigate it further because due to #628 there was high risk of breaking the neck.

pattacini, 2018-04-14:
If we want to focus merely on `iKinGazeCtrl`, my educated guess is that the inertial unit is not providing the gaze controller with the correct information for a number of reasons such as freezes, low rates; see for example discussions in #585. This assumption holds with Bosch IMU; @julijenv can you confirm the IMU vendor of this robot?

I would try out to disable the IMU anyway:
1. If not already done, copy out the configuration file in the user path of the computer where `iKinGazeCtrl` is supposed to be running: `yarp-config context --import iKinGazeCtrl`.
1. Open the configuration file in the user path `~/.local/share/yarp/contexts/iKinGazeCtrl/config.ini`.
1. Edit/fill in the file with the following option:
    ```ini
    [imu]
    mode off
    ```
1. Launch `iKinGazeCtrl` without command-line parameters from within the console and inspect the printed messages to check whether:
    1. The intended configuration file is loaded correctly: `||| found ${HOME}/.local/share/yarp/contexts/iKinGazeCtrl/config.ini`.
    1. The IMU is not employed in the computation: `[WARNING]IMU data will be not received/used`.

lornat75, 2018-04-16:
@karinne-ics-tumkarinne-ics-tum

pattacini, 2018-04-16:
More on this:
the head is version 1.0, therefore the IMU is an Xsense, not a Bosch.
That said, I believe it is still worth trying to disable the IMU as described above.

karinne-ics-tum, 2018-04-16:
Thanks for the information, I will try to disable the IMU of the robot.

closed 2018-06-04 by pattacini