robotology/mesh-support #632: Calibrating the little fingers after re-wiring

pattacini, 2018-04-16:
@tasbolat1 commented on [Mon Apr 16 2018](https://github.com/robotology/QA/issues/286)

Hello, everyone!

I’ve recently changed a tendon wire. However, cannot able to re-calibration of it. What’s the fast and correct way of doing so?

---

@pattacini commented on [Mon Apr 16 2018](https://github.com/robotology/QA/issues/286#issuecomment-381613817)

Hi @tasbolat1 
For hardware-wise issues, it's preferable to post on https://github.com/robotology/icub-support.
Thus, I'll redirect this port therein.

julijenv, 2018-04-16:
Hi Tasbolat,
As explained on skype, normally you dont need to do that. BUT as you never experienced any rewiring, I would suggest you to just launch the yarprobotinterface WITH REB BUTTON PRESSED and doucle check that the value for the finger stretch is close to zero and as soon as you do spin the motor pulley CCW the fingers simultaneously (together) close and the encoder value increase. If yes, just close the yarprobotinterface, release the red button and launch the yarprobotinterface again (you may have to do a power cycle of the motor switch as well, I would i I were you). 
LEt us know

julijenv, 2018-04-17:
Hi @tasbolat1 , they are marker ad unconfigured because you launched the harp robot interface with the red button pressed so of course the motore didnt do anything and of course they didnt male the calibration so NOT CONFIGURED is written all aover the place. Please restart the harp robot interface andò tell me if everything is Albright andò calibrate correctly.