robotology/mesh-support #633: Right wrist cable broken

DanielePucci, 2018-04-16:
### Description of the failure
A cable of the wight wrist is broken. 

[image] 

### Detailed conditions and logs of the failure

While performing the red-ball demo, the robot accidentally hit its right wrist against the left hand, which induced the cable breakage.

CC
@spada1978 @julijenv

julijenv, 2018-04-18:
Hi @DanielePucci ,
I dont know if someone is around able to help you out with this - maybe @spada1978 .
if not in the worse scenario hetre is the code of the 2 cable at the level of the wrist that could be snapped.

6418 | RC_IIT_019_G_027 | 53mm | Assemblies | WRIST V2 TENDON
6419 | RC_IIT_019_G_028 | 45mm | Assemblies | WRIST V2 TENDON

There is a short one and a longer one, just retreve both pieces of the broken one and identify the dimension of the cable that need to be replaced.
thx in advance

julijenv, 2018-04-23:
Hi @DanielePucci and @vtikha ,
The robot is now up and running.
It has been changed:
IITCODE Qta Description:
6419 | RC_IIT_019_G_028 | 45mm | Assemblies | WRIST V2 TENDON

closed 2018-04-23 by julijenv