robotology/mesh-support #635: Board 20 (head) not found at startup

gabrielenava, 2018-04-18:
### Description of the failure

The robot cannot start. It is not possible to connect with the board 20 (that is the first board in the daisy chain). Maybe a cable is broken?

### Detailed conditions and logs of the failure

At startup, there is an error saying it is not possible to connect with BOARD 20. I attached the logger to this issue. `FirmwareUpdater` cannot discover any board.

 
[log_icub-head_yarprobotinterface_1685.txt](https://github.com/robotology/icub-support/files/1923968/log_icub-head_yarprobotinterface_1685.txt)

julijenv, 2018-04-23:
Hi @gabrielenava ,
My colleague Davide G. just told me that he found out the 3 stands off broken in the head and of course surely something goes wrong when walking or even moving just a bit the head. I propose to block the robot ASAP possible after the demo in order to change all the stads off, check also, both #561 and #605.

julijenv, 2018-04-23:
Hi @gabrielenava , for this moment I just glued one of the stanoff in order to block the translation movement.

julijenv, 2018-04-27:
Not sure 100%, but normally it has been fixed by changing the stand offs that were broken and make the board moving,and probably making false contacts in some cases.
it has bben used
IITcODe Qta Description

772 x 3 : Spacer, hexagonal, M2.5, M/F, h=25mm, teflon

closed 2018-04-27 by julijenv