robotology/mesh-support #637: yarprobotinterface errors after update

emilio-cartoni, 2018-04-19:
### Description of the failure
We have updated both yarp and icub-main on the laptop and on the pc104.
However, after the update we get errors when starting yarprobotinterface.
Log attached.

### Detailed conditions and logs of the failure
In the log, it complains about "fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support".

How can we update configuration files so that they run with latest yarp/icub-main release?
[yarprunlog_19_04_2018_17_57_34.log](https://github.com/robotology/icub-support/files/1928910/yarprunlog_19_04_2018_17_57_34.log)

gsaponaro, 2018-04-19:
Hi @emilio-cartoni , a couple of suggestions:
1) on the pc104, in addition to yarp and icub-main, make sure that the **icub-firmware-shared** repository is also up-to-date (related manual page: http://wiki.icub.org/wiki/Compilation_on_the_pc104)
2) the parameters `fullscalePWM` and `ampsToSensor` should be set as in here:
- https://github.com/robotology/robots-configuration/pull/25,
- https://github.com/robotology/robots-configuration/commit/87f9ae6716486788856a3d0cf6ff28d52055b333

julijenv, 2018-04-24:
Hi @emilio-cartoni ,
yes please do ask for a pull request on your repository.

emilio-cartoni, 2018-04-24:
Pull request asked, thanks for the support :)
https://github.com/robotology/robots-configuration/pull/65

closed 2018-04-24 by emilio-cartoni