robotology/mesh-support #639: iKart battery status cannot be retrieved

Tobias-Fischer, 2018-04-24:
Hi,
We are having issues in getting the battery status of the iKart.
So far, we have tried the following:
1) Charged the battery. The LED on the charger now shows a green light (it was red at the beginning, then changed orange, and now is green). So we suppose the battery itself is fine.
2) We tried running the `iKartBatteryManager` with parameters `--from /usr/local/src/robot/install/ICUBcontrib/share/ICUBcontrib/contexts/iKart/batterySerialPort.ini`. However, the output is just as follows (repeats all over):
```
[DEBUG]0 
[DEBUG]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
[INFO]
```
As far as I understand the source code (https://github.com/robotology/ikart/blob/master/src/iKartBatteryManager/main.cpp), the 0 at the beginning indicates that no data was received.
3) The USB serial device seems to be recognized okay in Ubuntu. Running `dmesg | grep tty` yields in:
```
icub@ikart:~$ dmesg | grep tty
[    0.000000] console [tty0] enabled
[    0.631838] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    0.653339] 0000:00:16.3: ttyS4 at I/O 0xf160 (irq = 17, base_baud = 115200) is a 16550A
[    0.705421] tty tty49: hash matches
[   15.587441] cdc_acm 5-2:1.0: ttyACM0: USB ACM device
[   15.591451] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB0
```
4) One more thing: in the iKart Wiki, there is a light on the battery board (http://wiki.icub.org/wiki/File:IKART_BATTERY_BOARD.jpg). This does not show up on our board. Is that normal?

We are not sure how to proceed from here. It would be great if you could give us a hint.

Thanks a lot and best wishes,
Tobi

julijenv, 2018-04-24:
any hints on that  @randaz81

Tobias-Fischer, 2018-05-01:
Ping @randaz81

maggia80, 2018-05-09:
could you try with a terminal from Ubuntu (like gtkterm) with baudrate 57600 and no parity and 1 stop bit if you receive any chars?

ciao,
Marco

Tobias-Fischer, 2018-06-27:
Hi @maggia80,
Sorry for the delay. It seems that we don't receive any chars. Would you know why this could be?

[image]

Thanks,
Tobias
/cc @umgnunes @Twarz

maggia80, 2018-06-27:
I think that there could be an HW issue then. Could you check the cable connection on the BCS board?

[image]

Tobias-Fischer, 2018-06-28:
Hi @maggia80,
Our cable looks different - only the orange and yellow cable are going to the connector, but not the black one. However, we can see that there is a black cable that is not going to the connector. 

[image]

Do we need to put that one in the connector or is it supposed to work with just the yellow and orange cable?

Best, Tobi

maggia80, 2018-06-28:
Dear Tobi,

It is important that yellow and orange wires are properly connected and from what I can see they are good. Therefore the issue is on the BCS board. The only thing you can do is to change it. Let me check with @julijenv how we can arrange it.

Tobias-Fischer, 2018-06-28:
Hi @maggia80,
Thanks a lot, that would be very helpful. We have another delivery that is due to be shipped to us - maybe we can combine the shipments @julijenv?

Best, Tobi

Tobias-Fischer, 2019-10-08:
Dear @maggia80 @julijenv,
Did you happen to find a BCS board?

Best, Tobi

traversaro, 2019-12-07:
cc @maggia80 @julijenv

maggia80, 2019-12-09:
deeply sorry @Tobias-Fischer, we don't have BCS boards anymore and there are components mounted on that board that are obsolete. We should design a new one, but for the time being, it's not an activity that we are going to schedule.

pattacini, 2019-12-10:
Closing for now.

closed 2019-12-10 by pattacini

Tobias-Fischer, 2019-12-11:
/cc @umgnunes @yiannisdemiris