robotology/mesh-support #640: Unable to connect to YARP ports from MATLAB

vibhoraggarwal, 2018-04-24:
### Description of the failure
After switching on `icub-head` successfully, I run the `yarprobotinterface` from the YARP Manager. Even after starting successfully, the ports on the robot are not connected.  
- As I press the `Refresh Status` on YARP Manager, the `yarprobotinterface` stops. 
- I tried using `yarp clean --timeout 0.1` from terminal and then checked `yarp name list`. All the ports previously open by `yarprobotinterface` were turned off.

Following are the error messages that I am getting in **MATLAB** after running the model for 'l_ankle_pitch' joint:
### Detailed conditions and logs of the failure
```[[01;31mERROR[00m]Problem connecting to /icub/left_leg/rpc:i, is the remote device available?
[[01;31mERROR[00m]Problem connecting to /icub/left_leg/command:i, is the remote device available?
[[01;31mERROR[00m]*** Extended port /icub/left_leg/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility behaviour
[[01;33mWARNING[00m]Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
[[01;31mERROR[00m]RemoteControlBoardRemapper: error opening remote_controlboard with remote " /icub/left_leg ", opening the device failed.
cc: @gabrielenava

traversaro, 2018-04-24:
From 
> As I press the Refresh Status on YARP Manager, the yarprobotinterface stops
I guess that the yarprobotinterface is actually failing. Can you check on the `yarplogger` what is the output of the yarprobotinterface?

julijenv, 2018-04-24:
Hi @traversaro ,
I just assisted to the calibration of the robot and afterwards, the robot is still controlled but the yarprobotInterface on yarpmanager it just looks like it is not responding - there is the red cross nest to it. ( everything comes down to the fact that @vibhoraggarwal  was using iCubStratupYoga from the yarpmanager of the iCubGenova02 and I advised him to use the normal iCubStartup, but still the question remains on why the iCubStartupYoga dont work anymore).

traversaro, 2018-04-24:
> I just assisted to the calibration of the robot and afterwards, the robot is still controlled but the yarprobotInterface on yarpmanager it just looks like it is not responding

This seems to be a crash related to the additional wholeBodyDynamics device loaded during `iCubStartupYoga`. I suspect `yarp`, `icub-main` and all the other software have been recently updated and recompiled, while the `codyco-superbuild` (I am afraid we still need to migrate to the `robotology-superbuild` still need to be updated and recompiled).

vibhoraggarwal, 2018-04-24:
We changed the xml file used by changing **yarprobotinterface.ini** file in icub-head, with text in it, from `icub_no_arm.xml` to `icub_all.xml`. Now, the **yarp ports are readable**. Thanks @julijenv 

But, there is a problem. Matlab has stopped functioning and showing an error. I suspect, the error is related to YARP and not with Matlab; when I tried to run the model, Matlab, when it reached at a point it had to start working on robot, **failed** [(can be seen by clicking on this link)](https://user-images.githubusercontent.com/19661841/39199453-c57c96e8-47e9-11e8-8628-aa7376ab8830.png)
Following is a log from the matlab_crash_dump file, which I have also attached. Starting line shows some relation to YARP which I have never seen before?
[matlab_crush_dump.txt](https://github.com/robotology/icub-support/files/1943632/matlab_crush_dump.txt)

P.S: I already tried to work on using the solution mentioned (>>)[here](https://in.mathworks.com/support/bugreports/1297894) , but it does not work.

```Stack Trace (from fault):
[  0] 0x00007f53b81ccdbe /usr/local/src/robot/yarp/build/lib/libYARP_dev.so.1+01400254
[  1] 0x00007f53b8134bfc /usr/local/src/robot/yarp/build/lib/libYARP_dev.so.1+00777212
[  2] 0x00007f53b8919a94 /usr/local/src/robot/codyco-superbuild/build_icub24/install/mex/WBToolbox.mexa64+00334484
[  3] 0x00007f53b88e5ba4 /usr/local/src/robot/codyco-superbuild/build_icub24/install/mex/WBToolbox.mexa64+00121764
[  4] 0x00007f541f4365eb /usr/local/MATLAB/R2016b/bin/glnxa64/libmwsimulink.so+27583979
[  5] 0x00007f541f3ea343 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwsimulink.so+27272003

traversaro, 2018-04-24:
I suspect that also this crash is due to the fact that the `wb-toolbox` contained in the codyco-superbuild was not recompiled after the yarp update.

vibhoraggarwal, 2018-04-26:
I need to take measured torque data from the robot in `Torque control Mode`; only `yarprobotinterface` in the **iCubStartupYoga** supports the Torque Control Mode, so the problem still stands. 
@traversaro Although you helped in updating the `codyco-superbuild` repository on `icub-head` and `icub24`, but the problem that was mentioned in the [description](https://github.com/robotology/icub-support/issues/640#issue-317237748) still stands.

vibhoraggarwal, 2018-04-27:
The issue is now solved. 
There were some changes, done by @traversaro, in the `devel` branch of `../codyco-superbuild/main/codyco-modules` which were to be compiled on the **icub-head** before running `yarpmanager`
@julijenv Just for the clarification about `iCubStartupYoga` and `iCubStartup` about which we were discussing on the first day of problem: `iCubStartup` does not allow us to run in the TorqueControlMode(), which is allowed in `iCubStartupYoga`. Thus, we use `iCubStartupYoga`.
Thanks @traversaro and @julijenv for your help in solving the problem.
I am **closing** the issue now.

closed 2018-04-27 by vibhoraggarwal