robotology/mesh-support #642: Right shoulder cable loose

hu-yue, 2018-04-30:
### Description of the failure
The shoulder of the right arm seem to be loose.

### Detailed conditions and logs of the failure
The robot has been previously repaired (on the right arm, not exactly sure if due to https://github.com/robotology/icub-support/issues/638), afterwards, we updated `yarp` to `devel` and migrated all the parameters and gains as per migration guide.
When testing the new configuration files, we noticed that the right shoulder might have a loose cable, given that the arm oscillates a lot when pushed. In comparison, the left arm seems fine.
Please see GIF below:

[image]

@S-Dafarra @gabrielenava FYI

DanielePucci, 2018-05-01:
CC
@julijenv

hu-yue, 2018-05-04:
After reverting YARP and icub-main to the older version we did not observe this issue, so it most probably is a problem related to the migration of the gains to `metric_units` and not a cable issue.
Closing.

closed 2018-05-04 by hu-yue