robotology/mesh-support #644: Communication issue between Boards

gabrielenava, 2018-05-02:
### Description of the failure

At startup, the right arm and the left arm wrist and hand don't start the calibration procedure. We restarted the robot several times. Once it happened that they started the calibration, but the joints went in hardware fault after some time.

### Detailed conditions and logs of the failure
According to the logger (attached to this issue), there is a CRC error on board 25 (left arm). There is an old issue where it seems that the same thing happened (https://github.com/robotology/icub-support/issues/510): the errors in the loggers are the same.

According to the old issue, there might be a communication issue between the Board 25 and Board 3, that may explain the failure.

[log_icub-head_yarprobotinterface_15131.txt](https://github.com/robotology/icub-support/files/1967649/log_icub-head_yarprobotinterface_15131.txt)

hu-yue, 2018-05-04:
Closing, as it did not happen again.

hu-yue, 2018-05-08:
It happened again. We attach the log: [log_icub-head_yarprobotinterface_4180.txt](https://github.com/robotology/icub-support/files/1984369/log_icub-head_yarprobotinterface_4180.txt)

It did not show up again after switching off and on again the motors.

hu-yue, 2018-05-08:
@julijenv it happenend again, but not at the beginning.
We were trying to launch the Yoga++ when we got connection errors and realised that the connection was lost. The robot remained "on", but `yarprobotinterface` was down. We launched `FirmwareUpdater` and only boards 1,20,21,22,24,25 were on while the rest were off.
Unfortunately the logger was already saturated when we realised this problem, we attach it anyways: 
[log_icub-head_yarprobotinterface_4503_again.txt](https://github.com/robotology/icub-support/files/1984540/log_icub-head_yarprobotinterface_4503_again.txt)

@julijenv

julijenv, 2018-05-08:
found a connector almost discnonected. Hope this will close this issue.

hu-yue, 2018-05-09:
Thanks @julijenv, so far it has not happened, closing this.

closed 2018-05-09 by hu-yue