robotology/mesh-support #649: Left elbow tendon broken.

dhgarcia, 2018-05-09:
### Description of the failure
Hi.
Tendon broke on the left elbow on the Plymouth iCub during a demo. 
Can you help us to find instructions to replace it.
We have plenty of replacement cables. I believe this would be cable 9?
We took a look at the wiki http://wiki.icub.org/wiki/Tendons_and_replacement but I don't understand from it how to replace it. 
Where can we find instructions?
Thank you.

### Detailed conditions and logs of the failure

[image]

julijenv, 2018-05-09:
Hi @dhgarcia ,
from the picture I cant understand which cable is broken - to me it looks like just theat the cable popped up of it s position.
please download the document ShoulderWiring.pdf from the webpage of the icub wiki that you linked me earlier.
if it is the most inside one it is cable 9, or else it is 10.
frpm ShoulderWiring.pdf you will be able to understand how to replace the cable logically.
let me know

dhgarcia, 2018-05-09:
Thank you @julijenv 
Looking at the diagrams, I think maybe one cable broke and the other popped out.
We are going to attempt to replace it tomorrow.
I will let you know, and contact you if we need more help.
Thank you.

closed 2018-05-21 by dhgarcia