robotology/mesh-support #651: Battery dummy bad contact

hu-yue, 2018-05-10:
### Description of the failure

The dummy of the battery sometimes loses contact and the robot shuts down.

### Detailed conditions and logs of the failure

After switching the battery with the dummy, we realised that the dummy is losing contact sometimes, as we experienced a few times that the motors shut down due to lost contact of the dummy.
We saw that we need to push the dummy to establish the contact, as in the following GIF:
[image]

We saw later that the pins of the dummy are a bit pressed and might be the reason for the lost contact, we tried to straight them as in the following picture:
[image]

Still have to test whether this workaround solves the issue.

@julijenv @maggia80

hu-yue, 2018-08-16:
It seems that the leds on top are now just broken, since they just don't turn on at all but the robot can be switched on without problems.

DanielePucci, 2018-08-19:
We shall fine a more definite solution for this problem. What do you think @maggia80 @MrAndrea ?

MrAndrea, 2018-08-21:
We have to rethink the connection between the battery and the robot, and make it more robust.

kouroshD, 2019-10-03:
We had a similar issue in https://github.com/robotology/icub-tech-support/issues/779 , and to fix the connection problem, now we are using plastic screws.

closed 2019-10-03 by kouroshD