robotology/mesh-support #652: The robot went in hardware fault while torque controlled

S-Dafarra, 2018-05-14:
### Description of the failure
While performing the Yoga, the robot fell and several parts of the robot got switched off. In particular only the left arm and the right arm wrist were still up and running.

### Detailed conditions and logs of the failure
Here attached a log:
[logFaultYoga.txt](https://github.com/robotology/icub-support/files/1999582/logFaultYoga.txt)

I tried to connect to the boards later but the FirmwareUpdate **was able** to see all the boards.

cc @julijenv @DanielePucci @hu-yue @gabrielenava @GiulioRomualdi @maggia80

valegagge, 2018-05-14:
@S-Dafarra Have you already fixed this issue?

S-Dafarra, 2018-05-14:
@maggia80 is currently trying to fixing it. It seems to be a connection problem either between the ``eb1`` and the ``eb24`` or between the ``eb24`` and the ``eb25``. Can you confirm?

valegagge, 2018-05-14:
@S-Dafarra I read both logs and it seems that the problem is due to the ethernet cable that connects mc4plus with address 10.0.1.24 (eb24)  with mc4plus (10.0.1.25) eb25. Both boards are on left forearm.

Moreover, please ask to @julijenv to check the mais on left arm because it is often in error state.
(Probably the can cable in partially broken???....)

valegagge, 2018-05-14:
About mais error.... i think the problem is due to the connection of mais to eb26.

maggia80, 2018-05-14:
we checked the connection between MAIS and EB26 and EB25. The cables and crimps were good. 
We checked also ethernet connection between EB1 and EB24. EB25 and EB26. We crimped some wires between EB24 and EB25, but they weren't unconnected. We did it anyway to be sure the crimps were well done.

S-Dafarra, 2019-10-03:
I think this issue can be considered solved. We will open another issue in case it occurs again.

closed 2019-10-03 by S-Dafarra