robotology/mesh-support #655: iCubErzelli01:  head-eb21-j2_5 communication issues

ddetommaso, 2018-05-17:
### Description of the failure

The robot head stops randomly during the requested motion because the board head-eb21-j2_5 seems to have  communication problems with the PC104. The issue occurs the first time only after some minutes of use (around 30) starting from the robot off. After the first occurrence it occurs systematically, so we cannot run neither a experiment entirely.

### Detailed conditions and logs of the failure

I performed a test in which only the head is enabled (legs, arms, torso and skin are disabled). The test consists of a periodic task in which I send requests to look at 3D points using the iKinGazeCtrl.

Please attached you can find the logfile and a picture showing the network dropping.

[image]

[log_pc104_yarprobotinterface_2373.txt](https://github.com/robotology/icub-support/files/2013589/log_pc104_yarprobotinterface_2373.txt)

julijenv, 2018-05-17:
Hi @ddetommaso ,
looks like the connector entering inside the board EB21, has a bad connection.
you may have to ask for a recrimp.
I wont be in IIT before monday.
tell me if you are able to fetch someone to help you, before.

ddetommaso, 2018-05-18:
Hi @julijenv,

thank you for your feedback! I asked Davide Delle Piane from EDL and he recrimped the connectors E20 and E20 entering in the board EB21. Now I'm testing the robot. I'll update you later.

ddetommaso, 2018-05-18:
Hi @julijenv,

unfortunately the issue happens again after a couple of hours of using. Could you arrage a visit here in Erzelli next week?

Thanks

julijenv, 2018-05-21:
hi @ddetommaso ,
send me a brand new log and please entire log, thx in advance

julijenv, 2018-05-29:
any news, since last week @ddetommaso

ddetommaso, 2018-05-29:
hi @julijenv ,

we're still testing. So far, after the replacing of the board EB20, the issue seems gone.
I'll post here the results of the next tests.

ddetommaso, 2018-05-30:
Hi @julijenv ,

so far no errors related to the boards from yarprobotinterface. But, after around one hour working well, I got sistematically errors from the iKinGazeCtrl. It seems that the controller finds a "Communication Timeout" and goes in Suspend mode. Moreover I can clearly see a network drop accordingly with the timeout.

How can I check if the issue is hardware related ?

ddetommaso, 2018-05-30:
Just following my prev msg, I did the following test.

Once the "Communication Timeout" error occurred (after around one hour testing) I turned off both the robot (pc104 and motors) and the controlling PC. Then, I turned on all again after a couple of minutes.

I executed the test again and this time I got the errors from the boards (after several network drops).

Please attached you can find the logfile.
[log_icub-head_yarprobotinterface_1498.txt](https://github.com/robotology/icub-support/files/2054887/log_icub-head_yarprobotinterface_1498.txt)

julijenv, 2018-05-31:
Hi @ddetommaso ,
after an hour of testing, could you try to understand if the motors are warm, hot or just cold as started?
The errors show basivcally the same problems tha we have before changing the board.
some link goes down and uo in between board 20 and 21.
I doubt that it is the link in between them so I guess it is the 20 that starts messing around.
I dont know what to say. I am in Trento until monday. I cant help you until there. Can you make some pause every tot minuti in order to see if you can continue, withour the fault error popp up.

maggia80, 2018-06-04:
Hi @ddetommaso  and @julijenv may I suggest you to do some tests?

1)  could you try to run only the upperbody (disabling the board from robotinterface) or only the head if you are not using the other part of the robot? I would like to see what happen if we reduce the network traffic. 
Leave the robot running for one hour or more

2) Remove the back black cover of the head, enabling all the boards from robotinterface. Leave the robot running for one hour or more

3) Change the network connection sequence. Now the connection is PC104 -> EB20 -> EB21. I would like to change the sequence in: PC104 -> EB21 -> EB20. To do that we need @julijenv  

We have seen from the log files that the board EB20 resets, before and after replacing it. Therefore we want to understand why it happens. We suspect: overheating, power issue on board EB20.

ddetommaso, 2018-06-04:
Hi @maggia80  and @julijenv ,

thank you for the feedback, I was just testing what @maggia80 wrote, but this time the issue occurred only after 10 minutes from the first start. 

From yarprobotinterface interface I selected only the head, all other parts disabled. When the issue occurred I could clearly see network drops as well.

Please attached you can find a fresh log.

[log_pc104_yarprobotinterface_1752.txt](https://github.com/robotology/icub-support/files/2067697/log_pc104_yarprobotinterface_1752.txt)

Thank you

maggia80, 2018-06-04:
Thanks @ddetommaso  for the test. 

Does the head move during the test? Especially the neck. 

Could you perform test 2)? 

thanks,
Marco

ddetommaso, 2018-06-04:
Hi @maggia80 ,

the test consists of a periodic task in which the same sequence of movements are generated by the gaze controller, so the head (neck + eyes) are moving periodically.

Yes, I performed the test 2. The issue appeared after a couple of minutes from the start. I attach a log file of this condition as well.

[log_pc104_yarprobotinterface_1577.txt](https://github.com/robotology/icub-support/files/2068167/log_pc104_yarprobotinterface_1577.txt)

When the issue occurs, in both cases, the neck continues moving but only the YAW, in fact joints J0 and J1 stop working (from yarprobotinterface changing the controllin mode they appear "not configured").

julijenv, 2018-06-04:
Hi @ddetommaso ,
 @maggia80  suggested you to test to raise the tension up to 15V on the powersupply for the head and increase by consequence the OverVoltage of this power supply up to 17V.
Let us know,

ddetommaso, 2018-06-04:
Hi @julijenv  and @maggia80 ,

I changed those values on the power supply and now the behavior it's different.
This time the robot freeze for a bit during the movement but it continues working, the board that before was going down now it remains up. Again I can see that for each freeze of the movement there is a network drop.

maggia80, 2018-06-04:
Hi @ddetommaso,

Ok, then the problem of the reset may be related to a voltage drop. You can leave it at 15V.    
I have checked the log file and I don't see any issue related to packets lost or CRC errors.   
The network drop is on the 100Mb network? Or in the Gb network between the PC104 and icubs4hri?
Because if you see a drop on the Gb ethernet it may be related to the cable from the head to the back of the robot.

Could you try to run _iperf_  from the pc104 to the icubs4hri? This tool will check the bandwidth. 
Could you check if there are CRC errors in the Gb network? Using Wireshark or similar tools

ddetommaso, 2018-06-04:
Hi @maggia80 ,

the network drops were related to the Gb network between PC104 and icubs4hri.

The iperf tests between pc104 and icubs4hri confirmed that the bandwidth is 1Gb. With wireshark I couldn't see errors during the iperf tests, but after I was checking the interface statistics and I could see packet errors on the ifaces of both PCs.

[image]
[image]

maggia80, 2018-06-05:
Thanks @ddetommaso  for the tests,

@julijenv  you should do two things:

1) Bring with you a cable for connecting directly the COM-Express on the head with the patch in order to skip the part of the ethernet cable passing through the robot. If the problem disappears we need to check the ETH cable passing through the robot.

maggia80, 2018-06-05:
2. Check with the oscilloscope the voltage on the MC4-PLUS when the robot is switched on. We should have 12V at least.

julijenv, 2018-06-08:
TEST1: attach the extension cord with a brand new one BUT the problem was still occuring.
TEST2: We plug directlty to the board the ethernet cable provided with the board and at that point we couldnt notice any problem (we didint test for hours though)
TEST3: We cut 5cm of untwisted and unshieldded ethernet cable and I recrim,p the connector up to the board. After some tests we could notice no packets loses or network drop, and we wtill performing test to asset the good behaviour of the robot.
If the problem occurs again, the only thing that will be to do is replace the ethernet cable from the back cavoer of iCub up to the head.

julijenv, 2018-06-08:
Unfortunately before leaving some drophave been spotted, we then stressed everywhere the network cable , and finally with a lot of stress the neck cable finalaly released the truth,
the ethernet cable is somehow weaked at the level of the neck, it will be repaired on monday

julijenv, 2018-06-14:
hi @ddetommaso ,
can we consider this issue closed?!?! did you test the neck and the bandwith, did u succeed to have the yarpview to work again? let me know

ddetommaso, 2018-06-15:
Hi @julijenv ,

after the replacement of the eth cable in the back the issue seems gone!
Thanks a lot

closed 2018-06-20 by maggia80