robotology/mesh-support #658: yarprobotinterface segmentation fault after update

pi-q, 2018-05-18:
When launching yarprobotinterface on PC104, a segmentation fault occurs. No calibration and no startup of the robot happen.

We updated yarp, icub-main, codyco-superbuild, icub-firmware-build. We also updated the eth boards with FirmwareUpdater, to their latest versions (as far as we know, we successfully did it for mais, strain, skin and 2FOC boards).

We attach here the log obtained when running yarprobotinterface, and also screenshots of FirmwareUpdater. 

Are we missing anything?

[PC104_yarprobotinterface_log.txt](https://github.com/robotology/icub-support/files/2017703/PC104_yarprobotinterface_log.txt)
[image]
[image]

[image]

valegagge, 2018-05-21:
hi @pi-q ,

Did you update also icub-firmware-shared repository and EMS boards?

pi-q, 2018-05-21:
Hi @valegagge,

As far as I know, we did. EMS boards are now version 3.20. I can double check tomorrow that we successfully updated icub-firmware-shared and let you know.

Anything else I could check?

valegagge, 2018-05-21:
3.20 is the correct version of ems. 

Could you check also robots-configuration repository, please?

pi-q, 2018-05-22:
Yes, both icub-firmware-shared and robots-configuration have been updated

pi-q, 2018-05-22:
I have a doubt about updating the CFW2 boards, though. I don't think we updated them, since I am not too sure how to go about it. Could the problem be coming from there?

julijenv, 2018-05-22:
hi @pi-q ,
please open Firmware updater and flash the boards BLLID 1 with the last 2BLLDC.1.62.out.S updated file from the repository firmware-build
as well as the MC4 boards respectively with ID 2 - 4DC.2.15.out.S.
C if it works better

pi-q, 2018-05-22:
Hi @julijenv,
Same result after flashing the BLL and MC4 boards. 
I was looking at ETH board 10.0.1.11 though (see in the pictures above), and most of the mtb boards show to be in canBootloader, is this something?

valegagge, 2018-05-23:
hi @pi-q,

if the MC4 boards on ETH board 10.0.1.2 and 10.0.1.4 have build version 87 it is not necessary. 
Please pay attention to which file use to update MC4 boards, each of then needs a specific firmware that differs from the others.

Maybe, to investigate better this fault you can configure only one part of robot per time. For example you can run yarprobotinterface only on right arm using only motion control devices.

julijenv, 2018-05-23:
Hi @pi-q,
please try to catch me on skype tomorrow afternoon - my skype-id : icub.support.
C U maybe 2moro

marcoaccame, 2018-05-25:
Hi, 
I have noticed that the mc4 boards below the 10.0.1.2 and 10.0.1.4 have versions 1.25.87 / 1.40.87 and 1.48.87.

Well, i think that they should have the versions **2**.x.87 instead.

Thus, please update in the following way:
- 10.0.1.[2, 4]:CAN1:3   use file 4DC.2.19.out.S
- 10.0.1.[2, 4]:CAN1:5   use file 4DC.2.28.out.S
- 10.0.1.[2, 4]:CAN1:7   use file 4DC.2.30.out.S

About the mtb boards which are in bootloader state: try to update them one by one and not all together.

However, i don't think that those wrong FW on mc4 / mtb boards can explain the crash.

We can try to run yarprobotinterface in master mode on our robot iCubgenova02 (the purple one) which is the most  similar to the Nancy01. That could help identify possible bugs.

One more request: can you please try to run only one FT sensor (not motion or other)?

The xml file is something like  that:

```
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">

<robot name="iCubNancy01" build="1" portprefix="icub">
    <params file="hardware/electronics/pc104.xml" />

    <!-- ANALOG SENSOR FT -->
    <devices file="wrappers/FT/left_arm-FT_wrapper.xml" /> 
<!--    
    <devices file="wrappers/FT/right_arm-FT_wrapper.xml" />
    <devices file="wrappers/FT/right_leg-FT_wrapper.xml" /> 
    <devices file="wrappers/FT/left_leg-FT_wrapper.xml" />
    <devices file="wrappers/FT/right_foot-FT_wrapper.xml" /> 
    <devices file="wrappers/FT/left_foot-FT_wrapper.xml" />
  -->
  
    <devices file="hardware/FT/left_arm-eb1-j0_3-strain.xml" />
    
<!--    
    <devices file="hardware/FT/right_arm-eb3-j0_3-strain.xml" />
    <devices file="hardware/FT/left_leg-eb6-j0_3-strain.xml" /> 
    <devices file="hardware/FT/left_leg-eb7-j4_5-strain.xml" />
    <devices file="hardware/FT/right_leg-eb8-j0_3-strain.xml" /> 
    <devices file="hardware/FT/right_leg-eb9-j4_5-strain.xml" />
  -->

</robot> 

```

Thanks.

marcoaccame, 2018-05-25:
I forgot to write: the FT only mode is one of the most simple modes. It can be the starting point for a heavier debug.

pi-q, 2018-05-25:
Hi @marcoaccame.

Thank you for all that information! 

I updated the mc4 boards following your instructions, and the mtb boards are all fixed. Like you predicted, it did not solve the problem.

Then I tried running yarprobotinterface with just one FTsensor, with the xml file as you specified. The result is the same:
```
[WARNING]in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
[DEBUG]eth::parser::print(boardData) for BOARD left_arm-eb1-j0_3 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES: 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.1 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  ems4 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS: 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  left_arm-eb1-j0_3 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60 
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.1 (left_arm-eb1-j0_3) with timeout = 0.02 sec and period of missing report = 60 sec 
Segmentation fault
```

marcoaccame, 2018-05-25:
Hi @pi-q,
the test on iCubGenova02 w/ master was OK: I ran yarprobotinterface w/ just one FTsensor and there was no crash. That is a good starting point.

I suspect that there is something weird in the system settings of your robot. From Monday I will help you and @julijenv to solve the problem.

pi-q, 2018-05-25:
Hi @marcoaccame,

That's great! I will be available next Monday try to figure it out.

marcoaccame, 2018-05-28:
Hi @pi-q,

here are some more requests:
1. run the FTsensor-only test w/ the motors turned off
2. run the FTsensor-only test w/ motors turned on as you did on Friday. 
3. power off and then on and run the FTsensor-only test w/ the motors turned on but onto a not existing IP address (see note-1 for details).

Of them all, please keep the logs and sent them to me.

From these tests I want to see if the crash happens  also in absence of every EMS boards (test No. 1) and when only the target one is missing (test No. 3). I need also the complete log of the the normal case (test No. 2).

note-1.
To use a not-existing IP address (e.g., 10.0.1.99) for the FTsensor-only test, edit file `left_arm-eb1-j0-3-eln.xml` so that you have:

```        
<group name="ETH_BOARD_PROPERTIES">
            <param name="IpAddress">                10.0.1.99                </param>
```

Thank you.

pi-q, 2018-05-28:
Hi @marcoaccame,

Thank you for these instructions! I attach the logs resulting from each of these 3 tests.

Tests 1 and 2 gave the same result as before, and test 3 actually showed an error before the segmentation fault:
```
 ...
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type ems4 with IP =  10.0.1.99 
[INFO]The embObjStrain device using BOARD left_arm-eb1-j0_3  w/ IP 10.0.1.99 has the following service config: 
[INFO]- acquisitionrate = 10 
[INFO]- useCalibration = true 
[INFO]- STRAIN of type strain named id_l_upper_arm_strain @ CAN2:13 with required protocol version = (1, 0) and required firmware version = (0, 0, 0) 
[ERROR]theNVmanager::Impl::transceiverZ() cannot obtain from TheEthManager a EthResource * for IP = 10.0.1.99
Segmentation fault
```

[log_FTsensory-only_motorsOff.txt](https://github.com/robotology/icub-support/files/2044233/log_FTsensory-only_motorsOff.txt)
[log_FTsensory-only_motorsOn.txt](https://github.com/robotology/icub-support/files/2044234/log_FTsensory-only_motorsOn.txt)
[log_FTsensory-only_motorsOn_IPnotExisting.txt](https://github.com/robotology/icub-support/files/2044235/log_FTsensory-only_motorsOn_IPnotExisting.txt)

marcoaccame, 2018-05-28:
Hi @pi-q,

the error is understandable: the IP .99 is beyond the allowed by the implementation. 
```
[ERROR]theNVmanager::Impl::transceiverZ() cannot obtain from TheEthManager a 
EthResource * for IP = 10.0.1.99
```
My mistake. Use 10.0.1.30 instead.

pi-q, 2018-05-28:
Hi @marcoaccame,

Here is the new log from test 3, with 10.0.1.30, it seems to me that it is giving the same result as the other tests.

[log_FTsensory-only_motorsOn_IPnotExisting.txt](https://github.com/robotology/icub-support/files/2044406/log_FTsensory-only_motorsOn_IPnotExisting.txt)

marcoaccame, 2018-05-28:
Yes, I wanted to see that.

Here is what I think.

The problems is entirely inside PC104 and not in the FW because we saw the same kind of failure also with the ETH boards switched off. 

It is good news because it is much easier to debug.

We need to get again a skype session w/ @julijenv and we shall see: called xml files, old SW installations, focussed debug prints which we can put in the source code to spot the exact failure point.

julijenv, 2018-05-28:
Hi @marcoaccame  and @pi-q ,
for me it is whenever you can, please try to agree on when both of you are available.
let me know

marcoaccame, 2018-05-29:
In the yesterday's session (@pi-q and myself):

1. We updated yarp, icub-main, icub-firmware-shared, icub-firmware, etc. We updated the packages according to the wiki. And we deleted the build and build-pc104 dirs and recompiled everything.

2. We saw that yarprobotinterface crashes when there is a copy from a std::string to another std::string in file ethBoards.cpp. The same instruction works on the on the iCubGenova02/04 robots. If we remove this instruction the FTsensor-only test runs smoothly.

```
bool eth::EthBoards::add(eth::AbstractEthResource* res)
{
    ...
    // it crashes. any use of std::string causes the crash
    LUT[index].name = res->getProperties().boardnameString; 
    ...
}
```

3. We spotted that the Linux on PC104 is not updated. As a result, the compiler version is different (older) from the one used in iCubGenova02. May it be this the reason of the crash?  

In any case it is recommended:
- to update of the Linux on PC104 (http://wiki.icub.org/wiki/The_Linux_on_the_pc104),
- and then do a complete re-build w/ the new compiler.

traversaro, 2018-05-29:
Given what @marcoaccame commented, I guess this is a problem of the ABI mismatch for `std::string` between compilation units compiled with C++11 and compilation units compiled with C++98. This ABI mismatch is there in GCC <= 4, but it does not exist for GCC >= 5. 

For more on this problem: 
* https://developers.redhat.com/blog/2015/02/05/gcc5-and-the-c11-abi/
* https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_dual_abi.html 
* https://gcc.gnu.org/wiki/Cxx11AbiCompatibility

Note that this  occurred because we started using C++11 features in YARP headers, and we did this only because Debian Jessie is not a supported version of Debian in YARP releases anymore, see http://wiki.icub.org/wiki/YARP_Supported_Distributions .

julijenv, 2018-06-14:
Hi @InriaBrice , @pi-q , @dogoepp ,
any news?

InriaBrice, 2018-06-15:
It seems that the can CFW2<0> bll and mc4 have no can protocol version. When we use the Firmwareupdater it's written N/A when we clic on both of them. Before we had access to the eth maintenance option, but after recompiling everything, we can't anymore. 

Theses page doesn't help us :
 http://wiki.icub.org/wiki/Compilation_on_the_pc104
 http://wiki.icub.org/wiki/Linux:Installation_from_sources

Because the options have changed, (there is more option now + they switched by default to ON or OFF), we compared with the old installation version.

InriaBrice, 2018-06-15:
I tryed to update the firmware of theses two card using the automatic script, unless i failed with the ID it's giving this : 

canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 1 --firmware 2BLLDC.1.62.out.S
Initializing prompt version of canLoader...
Selecting canDeviceType cfw2can canDeviceNum 0, boardId 1, firmware 2BLLDC.1.62.out.S
Cfw2Can::open() returning error -2
yarpdev: ***ERROR*** driver <cfw2can> was found but could not open
[ERROR]no boards found 
[ERROR]check canbus cable, power supply connection etc.
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 2 --firmware 4DC.2.15.out.S
Initializing prompt version of canLoader...
Selecting canDeviceType cfw2can canDeviceNum 0, boardId 2, firmware 4DC.2.15.out.S
Cfw2Can::open() returning error -2
yarpdev: ***ERROR*** driver <cfw2can> was found but could not open
[ERROR]no boards found 
[ERROR]check canbus cable, power supply connection etc.
cfw2can 0 1 2BLLDC.1.62.out.S --> Error: ERR_NO_BOARDS_FOUND
cfw2can 0 2 4DC.2.15.out.S --> Error: ERR_NO_BOARDS_FOUND

When i open Firmwareupdater it's written :  

can0:1 bll
can0:2 mc4

[image]

marcoaccame, 2018-06-18:
The boards on CFW2<0>/<9> already have the latest FW version and don't need to be updated.
I will look however at why there is N/A in the right window.

About the automatic script ... I don't know why yarpdev has such a failure. But it is not worth to find out because this script will be made obsolete in a few days when we release iCub version 1.12.

The new method uses the same underlying library as FirmwareUpdater, as it is already described in https://wiki.icub.org/wiki/Firmware. 

But for now, if you need to view/load FW: please use FirmwareUpdater in GUI mode.

julijenv, 2018-06-19:
Hi @pi-q , @InriaBrice , @dogoepp  and @serena-ivaldi ,
could you tell me if the yarprobootinterface starts now??? if not could you please send us a log file to understand the failure, thx in advance

InriaBrice, 2018-06-25:
Hi everyone,

So i found what was wrong, and i'll explain what i did : 

I recompiled everything from scratch, with the Correct setting for pc104 and for the remote pc respecting this : 

http://wiki.icub.org/wiki/Compilation_on_the_pc104
http://wiki.icub.org/wiki/Linux:Installation_from_sources

Then i checkout on devel branch of icub-main otherwise it was not accepting YARP 3.0.0 version
I recompiled.

I updated all the firmware with the devel branch of icub-firmware-shared.
For mc4 i did it as marcoaccame said : 

    10.0.1.[2, 4]:CAN1:3 use file 4DC.2.19.out.S
    10.0.1.[2, 4]:CAN1:5 use file 4DC.2.28.out.S
    10.0.1.[2, 4]:CAN1:7 use file 4DC.2.30.out.S

Then it was still saying missing parameters : 

In the file general.xml i added
        <param name="MotioncontrolVersion"> 5 </param>
to the group
        <group name="GENERAL">
In the file of hardware/motorControl i added
       <param name="fbkControlUnits">     machine_units            </param>
       <param name="outputControlUnits"> machine_units </param>
 for the group
      <group name="POS_PID_DEFAULT">
      <group name="TRQ_PID_DEFAULT">
      <group name="2FOC_CUR_CONTROL">

Then it was complaining about the famous WRONG PROTOCOL VERSION and asking for version 1.6.
And i reading a bit more the error was looking for the configuration files of the service in hardware/motorControl.

So i just changed the version from 1.5 to 1.6 for the files that were needed : 

<group name="CANBOARDS">
                <param name="type">                 foc         </param>
                <group name="PROTOCOL">
                    <param name="major">            1           </param>
                    <param name="minor">            6           </param>
                </group>
                <group name="FIRMWARE">
                    <param name="major">            0           </param>
                    <param name="minor">            0           </param>
                    <param name="build">            0           </param>
                </group>
            </group>

After a reboot i launched yarpserver on pc104 and yarprobotinterface with the config file robots-configuration/iCubNancy01/yarprobotinterface.ini

Now the robot is moving and yarprobotinterface is not doing a segmentation fault anymore.

There is still few error, but i dunno what they mean.

Best regards

julijenv, 2018-06-25:
Hi @InriaBrice ,
Could you please attach a log file? maybe we could help you to resolve the problem, you cant understand?
cheers

InriaBrice, 2018-06-25:
Here is a piece of the several errors we have : 
```
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[DEBUG]Left_Arm_Calibrator : Calling calibrateJoint on joint  5 : type  2  with params:  750 -10 0 0 0 
[DEBUG]Left_Arm_Calibrator : set 3 j 5 : Calibrating... enc values AFTER calib:  0 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.06[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]cfw2can [0] printing boards infos:
None
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG] from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 1, time 4074s 843m 203u: (code 0x04000009, par16 0x000a par64 0x0000000000000000) -> DEBUG: hw fault2 detected: in par16 there are controlmodestatus in lsb and frame[1] in msb + .
[DEBUG] from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 2, time 4074s 843m 331u: (code 0x04000009, par16 0xa00a par64 0x0000000000000000) -> DEBUG: hw fault2 detected: in par16 there are controlmodestatus in lsb and frame[1] in msb + .
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG] from BOARD 10.0.1.4 (right_arm-eb4-j4_15), src LOCAL, adr 1, time 4075s 595m 99u: (code 0x04000009, par16 0x000a par64 0x0000000000000000) -> DEBUG: hw fault2 detected: in par16 there are controlmodestatus in lsb and frame[1] in msb + .
[DEBUG] from BOARD 10.0.1.4 (right_arm-eb4-j4_15), src LOCAL, adr 2, time 4075s 595m 227u: (code 0x04000009, par16 0xa00a par64 0x0000000000000000) -> DEBUG: hw fault2 detected: in par16 there are controlmodestatus in lsb and frame[1] in msb + .
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]Right_Arm_Calibrator : Timeout while calibrating  5 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]Right_Arm_Calibrator : set 3 : Calibration went wrong! Disabling axes and keeping safe pid limit
 
[ERROR]embObjMotionControl::checkRemoteControlModeStatus(1, 7103593) is unable to check the control mode of BOARD right_arm-eb4-j4_15 IP 10.0.1.4 because it is now in HW_FAULT
```

It is saying that the joints are in HW_FAULT status

InriaBrice, 2018-06-25:
Here after a reboot, and more longer one 
[log_after_reboot.txt](https://github.com/robotology/icub-support/files/2133854/log_after_reboot.txt)

julijenv, 2018-06-26:
Hi @InriaBrice ,
I cant see real ERROR messages, could you please try to run yarplogger and just display the red lines to double check what are the labeled ERROR messages, if any.
Could you tell me if the robot cablibrate correclty and when after ethe calibration phase, the robotmotorgui show some weird behaviour, some joint got UNCONFIGURED (grey insted of green) or RED,
let me know

InriaBrice, 2018-06-26:
I think that this hardware fault came from an issue that we have with the left hand.
Because it's saying MAIS error, and with what i am guessing it's about the hands.
We have a cable for index finger that is blocked. See picture.
[image]

Actually i think i just need to set the right parameters for the config files, the right server name on  pc104 and it should works.

I am already able to run yarpmanager-console with two iKinCartesianSolver with yarprobotinterface

julijenv, 2018-06-26:
Hi @InriaBrice ,
Could you please send me the screenshot of each arm on robotmotorgui after calibration,
thx in advance

InriaBrice, 2018-06-26:
I don't have access to GUI for now, i mean there is a lot of things that are still broken (including GUI). But when i can i'll send it.

Best

julijenv, 2018-06-26:
@marcoaccame , could you please try to understand why this message occured?
  

> [DEBUG]cfw2can [0] msg from board 2: MAIS error 100 

beacause on their robot can [0] is just the head and torso, the rest is ethernet related from what I recall

thx in advance

julijenv, 2018-06-26:
Hi @InriaBrice ,
ASA you got the GUI to work with just drop us a reply on the issue, thx in advance

marcoaccame, 2018-06-27:
Dear @InriaBrice, please **DO NOT** use the robot with any mixed master and devel branch.

Thus, having read from your previous posts what you checked out ... for now: **stop using the robot**.

If YARP's latest master 3.x version is not compatible with icub-main / icub-firmware-shared / robots-configuration etc in master branch there is a reason. It can be dangerous to attempt to fix it unless you really know what you are doing.

What you must do now is:
- check out latest 2.x release of YARP (https://github.com/robotology/yarp/releases/tag/v2.3.72.1)
- revert every other repository in master (icub-main, icub-firmware-shared, icub-firmware-build, robots-configuration).
- redo fw update of everything to latest master fw versions.

Please do it together with @julijenv

marcoaccame, 2018-06-27:
For use of YARP in iCub, see: https://github.com/robotology/QA/issues/300

InriaBrice, 2018-06-27:
Hi @marcoaccame , @julijenv 

I changed again to yarp 2.3.xx, aswell i changed the firmware.
There is some errors that ends in a segfault : 
```
[INFO]The embObjStrain device using BOARD right_leg-eb9-j4_5  w/ IP 10.0.1.9 has the following service config: 
[INFO]- acquisitionrate = 10 
[INFO]- useCalibration = true 
[INFO]- STRAIN of type strain named id_r_lower_leg_strain @ CAN2:1 with required protocol version = (1, 0) and required firmware version = (0, 0, 0) 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[DEBUG]cfw2can [0] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]cfw2can [0] printing boards infos:
None
[WARNING]cfw2can [0] have not heard from board 1 (channel 0) since 12.13 seconds
[WARNING]cfw2can [0] have not heard from board 1 (channel 1) since 12.13 seconds
[WARNING]cfw2can [0] have not heard from board 2 (channel 0) since 12.13 seconds
[WARNING]cfw2can [0] have not heard from board 2 (channel 1) since 12.13 seconds
[WARNING]cfw2can [0] have not heard from board 3 (channel 0) since 12.13 seconds
[WARNING]cfw2can [0] have not heard from board 3 (channel 1) since 12.13 seconds
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[DEBUG]cfw2can [0] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]cfw2can [0] printing boards infos:
None
[WARNING]cfw2can [0] have not heard from board 1 (channel 0) since 18.14 seconds
[WARNING]cfw2can [0] have not heard from board 1 (channel 1) since 18.14 seconds
[WARNING]cfw2can [0] have not heard from board 2 (channel 0) since 18.14 seconds
[WARNING]cfw2can [0] have not heard from board 2 (channel 1) since 18.14 seconds
[WARNING]cfw2can [0] have not heard from board 3 (channel 0) since 18.14 seconds
[WARNING]cfw2can [0] have not heard from board 3 (channel 1) since 18.14 seconds
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[ERROR]theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 
[DEBUG]cfw2can [0] msg from board 2: MAIS error 100 
[... cut ...]
[ERROR]theNVmanager::Impl::command() fails a wait() from IP 10.0.1.9 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult 
[ERROR]EthResource::serviceCommand() failed an acked activation request to BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 after 3 attempts each with waiting timeout of 0.5 seconds 
yarpdev: ***ERROR*** driver <embObjStrain> was found but could not open
[WARNING]Cannot open device right_leg-eb9-j4_5-strain 
[WARNING]Cannot open device right_leg-eb9-j4_5-strain
```
And there is still the MAIS 100 Error.

Best regards

julijenv, 2018-06-27:
HIi @InriaBrice ,
is it possible to get a full log and the the xml file launched by yarprobotinterface?
thx in advance,
on skype if ever eneeded

InriaBrice, 2018-06-27:
Sure, here it is
[log_yarp_2_3_72.txt](https://github.com/robotology/icub-support/files/2141738/log_yarp_2_3_72.txt)

And the full archive we used
[iCubNancy01.zip](https://github.com/robotology/icub-support/files/2141740/iCubNancy01.zip)

Best

julijenv, 2018-06-27:
Hi @InriaBrice ,
could youp lease try to checkout your repository from the repository robots-configuration, I noticed that some files have been changed. And to well understand it would be nice to understand if the online repository that should reflect the good behaviout of the robot before the upgrade/update, we need to test it.
let me know

InriaBrice, 2018-06-28:
Hi @julijenv 

The files inside are differents from the repository because we updated them. If we use the exact same files of the repository it say that we have the wrong version protocol on some board. (This came from the configuration files see above). And otherwise it give the same errors. We just changed the name from icub to iCubNancy01 for the YARP ports.

marcoaccame, 2018-06-28:
Hi, are you sure you have the correct FW on the CFW2:0 bus?

It should be as in the picture that @pi-q posted on May 18:

[image]

Can you please check w/ FirmwareUpdater? I suspect that in CAN0:2 you have a mc4 board w/ version 2.48.87 which is ... wrong and would explain the mais error.

julijenv, 2018-06-28:
HI @pi-q , @InriaBrice ,
Could we get in touch via skype to try to find out if what @marcoaccame is ok or not?
let me know, I ma right now behind my desktop for any session of skype if needed

InriaBrice, 2018-06-29:
Hi @marcoaccame , @julijenv 

i can today or next week for skype, setting the frimware of the mc4 board to 2.21.87 removed the MAIS error indeed. this is the new logs
[log.txt](https://github.com/robotology/icub-support/files/2148636/log.txt)

Best

marcoaccame, 2018-06-29:
I can see now only HW faults on both forearms. Can you send a printscreen of FirmwareUpdater with CAN1 of boards 10.0.1.2 and 1.0.1.4? Thanks.

InriaBrice, 2018-06-29:
[image]
 The firmware versions

InriaBrice, 2018-06-29:
[image]
[image]
Maybe this can help you too

InriaBrice, 2018-06-29:
[log_pc104_yarprobotinterface_julien.txt](https://github.com/robotology/icub-support/files/2149775/log_pc104_yarprobotinterface_julien.txt)
log we got with julien

InriaBrice, 2018-06-29:
Last log we got with @julijenv 
[log_pc104_julien.txt](https://github.com/robotology/icub-support/files/2150207/log_pc104_julien.txt)

InriaBrice, 2018-07-02:
Just a little update : 

We fixed the last friday with @julijenv  the robot, our issue was wrong drivers on the arms / hands. We needed version 1 because our robot is a bit different. 
I fixed the GUI also, now everything works fine. So we can close the issue.

Thanks for your help.
Best regards,

closed 2018-07-02 by InriaBrice