robotology/mesh-support #660: Yarprobotinterface fails with embObjMotionControl error for EB3

dcam0050, 2018-05-21:
### Description of the failure
`yarprobotinterface` fails during setting up of the robot. It seems like it cannot communicate with EB3 on 10.0.1.3

```
[WARNING]EthResource::serviceCommand() had a timeout of 0.5 secs when sending an activation request to BOARD EB3 with IP 10.0.1.3 
[ERROR]embObjMotionControl::open() fails to start mc service for board EB3 IP 10.0.1.3 : cannot continue 
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open
```

 or with its attached CAN boards

```[ERROR] from BOARD 10.0.1.3 (EB3) @ 180s 41m 341u: CAN discovery after 2328 ms has detected 4 missing eobrd_foc boards in CAN1:```

The following is the full detailed log:
[icub_error.log](https://github.com/robotology/icub-support/files/2023125/icub_error.log)

Any input on how I can debug this issue?

valegagge, 2018-05-22:
Hi @dcam0050 ,

the problem seems to be due to the lack of response of the MC4 CAN boards to the discovery made by the EMS. In details, during the startup phase, each ems verify that each CAN boards connected to is available with a "CAN dicovery" procedure.

In your case, none MC4 CAN board respoonds to this "CAN dicovery", so I think that maybe there is a problem to a CAN cable between ems and MC4 CAN boards.

To be sure of this hypotesis, please run FirmwareUpdater and check with it if the MC4 CAN boards are availble.
If you never have perform this task, please refer to section 4 of: 
https://github.com/robotology/icub-firmware-build/blob/master/ETH/FirmwareUpdater.readme.maintenance.of.eth.robots.txt

without update any firmware.

If the FirmwareUpdater can't find any MC4 CAN boards connected to ems with ip address 10.0.1.3 then the problem is due to a CAN cable broken.

julijenv, 2018-06-06:
Hi @dcam0050 , did u have good news, or  do u need me to investigate the problem.
let me know

julijenv, 2018-07-19:
Hi @dcam0050 , any news?!?!?!

dcam0050, 2018-08-23:
Hey everyone, sorry for going dead. The issue was occurring intermittently and we were not using the iCub that frequently. We are now planning to use it again and the issue is happening consistently.

I've used ethLoader to discover all eth boards and then launched canLoader to find all attached can boards. The following is what I get. It seems like there are too few can boards attached but I do not remember how much there were originally.

[image]

julijenv, 2018-08-23:
Hi @dcam0050 ,
from what I can see the EB3 give you back just the board attached to the CAN2 but no reply from boards on CAN1, (2, 2FOC boards) could you check the connection at the level of this board?
[image]
let us know

dcam0050, 2018-08-23:
In which part of the body is EB3 CAN 1 located?

dcam0050, 2018-08-23:
Found 2B0 and 2B1 on iCub's back using the Motor Board placement pdf and have removed the back plate and the heatsinks labelled 2B0 and 2B1. My plan is to remove and note all wires and connectors for the 2FOC boards and then start testing continuity on the cables connecting EB3 to the 2FOC boards. Would that be correct?

dcam0050, 2018-08-23:
[image]
[image]
[image]

julijenv, 2018-08-24:
Hi @dcam0050 ,
first of all, why did you remove the heatsink?!?!? I could have sent you the exact position where to look at!!!
Hope you have some heat condictive paste to restore the boards heatsink system.
First, could you just make sure that when you turn on the motors switch that the boards turn ON and behave the same way as the than boards 1B0 and 2B0, I mean all the blinking leds.
the 2 2FOC boards are connected via c, I double check today and the document I sent you yeterday was not readable enough to identify the connetors. so here is one document you may follow to check th econnection in between can1 of EMS3 and 2B0 and 2B1.
[iCub 3_3.0.4_5672_Logic.pdf](https://github.com/robotology/icub-support/files/2317599/iCub.3_3.0.4_5672_Logic.pdf)
if you go on the page rightarm logic you will be able to see that you need to check from board EMS3 on port 7 the labelled wire 3N1 which goes to P4 on 2B0 and later the connection in between 2B0 and 2B1 from P3 on 2B0 to P3 on 2B1 labelled 3N1. Check if those cables have some wires broken, esepcially the first one in between EMS and 2FOC.
Here some pictures of EMS and 2FOC boards bot side with the connector labelled on it, like that you can restore the heatsink palte on the 2 2 FOC.
[image]
[image]
[image]
[image]
let me know

dcam0050, 2018-08-24:
OK so I've put the heatsinks back on after cleaning and reapplying thermal paste and I start the motors.
I've put an unlisted video here https://youtu.be/wLXkeziAuk0

It seems like they boot the same but then I left it for longer and now they are at a point where the 2FOC boards on the left of icub (1B0 and 1B1) flash orange at a low frequency while the boards on the right side (2B0 and 2B1) flash green at a high frequency.

Not sure how good the picture of the current state has turned out but it's this one below.
[image]

I'm assuming high frequency green means still trying to establish communication with EB3 board which means I now need to switch off icub, disconnect 2B0 and check continuity of cable connecting 2B0 with EB3. Is this correct?

dcam0050, 2018-08-24:
Actually, I tried it again and 1B0 and 1B1 turned to low frequency orange because I was checking CAN devices through canLoader. When I checked CAN for 10.0.0.1, 1B0 and 1B1 turned low frequency orange and showed up. When I checked 10.0.0.3, 2B0 and 2B1 were unaffected and kept high-frequency green flashing.

I assume that still means I need to disconnect 2B0 and check comms with EB3 yes?

dcam0050, 2018-08-24:
Hey @julijenv. I removed the cable and tested for continuity between connectors and there was continuity on all wires. So I put it back in again and what do you know, all the CAN boards are visible now! I've closed iCub up, tried booting and it boots correctly now! I'll close this issue for now and reopen it should it still be intermittent. Thanks for your help!

dcam0050, 2018-11-19:
The CAN cable 3N1 between EB3 and 2FOC 2B0 is again working intermittently depending on how the wire is moved. We have some jst connectors to create a replacement cable but could you tell us the model number of the connectors (or their parameters) so we can replace like for like?

julijenv, 2018-11-20:
Hi @dcam0050 ,
here is the harness of the connector, please
[image]
the connector are MOLEX picoblade female 6 contacts(Housing, Picoblade, 6 contacts, 1 row, female, p.1.25mm, Straight, w to w / w to b, nolock, nokey, 1A)
[image]

the crimps are :Crimp terminal, Picoblade, female contact, AWG28-32
[image]
,hope this will help

closed 2019-10-12 by pattacini