robotology/mesh-support #664: CRC error from head board when launching yarprobotinterface

S-Dafarra, 2018-06-01:
### Description of the failure
The ``yarprobotinterface`` fails to start due to a CRC error:
```
24842	327,495569	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 36s 78m 671u: (code 0x06000002, par16 0x0001 par64 0x3000000000000015) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = P2, 1 = P3, 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
```

### Detailed conditions and logs of the failure
Here the boards that can be seen through the ``FirmwareUpdater``
[image]
 Here attached the two parts of the logger:
[log_icub-head_yarprobotinterface_1938.txt](https://github.com/robotology/icub-support/files/2062055/log_icub-head_yarprobotinterface_1938.txt)
[log_icub-head_yarprobotinterface_1939.txt](https://github.com/robotology/icub-support/files/2062056/log_icub-head_yarprobotinterface_1939.txt)

cc @DanielePucci @gabrielenava @marcoaccame @maggia80

S-Dafarra, 2018-06-01:
After few attempts, the robot started correctly. With @valegagge we thought it may be a "random" noise caused by some motors. Some more investigation may be necessary.

DanielePucci, 2018-06-01:
This *random* noise keeps occurring even on other robots (I remember that this CRC error occurred also on iCubGenova04, but I am not sure about the issue to point).

We shall understand the source of the problem in the furture

@maggia80 @marcoaccame

maggia80, 2018-06-04:
Looking at the log, I see CRC errors occurring on board EB20 in port ETH-OUTPUT (from EB20 to EB21). CRC errors to me should be related to wiring issues, therefore I will check the cable E20.

maggia80, 2018-06-04:
Regarding iCubGenova04 please add the logs of the errors here : #652 or #659

closed 2019-08-01 by julijenv