robotology/mesh-support #665: Oscillating gaze during demoRedBall

traversaro, 2018-06-03:
### Description of the failure
During the execution of the [`demoRedBall`](https://github.com/robotology/icub-basic-demos/tree/master/demoRedBall) the behavior of the eyes is quite oscillating even if the gaze target (the red ball) is not moving at all. 

### Detailed conditions and logs of the failure
This is the video of the behavior. Detailed logs will be provided once the robot will be back in the lab: 
[oscillating_gaze.mp4.zip](https://github.com/robotology/icub-support/files/2065771/oscillating_gaze.mp4.zip)

traversaro, 2018-06-03:
It seems that there seem some kind of "overshoot" in the eyes controller. The first debug will be to plot the desired and measured joint encoders and to check if there is anything strange there. 

@pattacini do you have any further insight?

pattacini, 2018-06-03:
Is IMU disabled?
It suffices to specify: `--imu::mode off`.

pattacini, 2018-06-04:
I believe it is more convenient to investigate the problem live when the system will be re-installed here in Genoa.

traversaro, 2018-06-04:
I agree.

kt10aan, 2018-08-22:
Do you have any updates on this? It happens to our robot too.

kt10aan, 2018-08-23:
@julijenv , we have no mechanical play at all.

julijenv, 2018-08-23:
Hi @kt10aan , as previously mentionned I just wanted to make sure that no play at the level of the eyes are present on the robot after that it finished calibrating. if no, so the problem is more due to control issue not hardware one so I will leave @traversaro , @pattacini  to find out where is the problem coming from.

kt10aan, 2018-08-23:
Here is a video https://www.dropbox.com/s/t32ndt59c7d9msi/IMG_0334.TRIM.MOV?dl=0
Notice that even when my head is steady, the robot's head moves in the pith axis bak and forth and also its eyes.

julijenv, 2018-08-23:
Hi @kt10aan ,
could you try to stay still in some position for at least 3 to 5s to understand what is wrong with the neck. To me it is the only thing that looks not perfectly ok but of course I'll wait from another video where you go from position to position with some rest in between them to see the behaviour of the head-eyes-neck movement.
thx in advance

kt10aan, 2018-08-28:
Here is a video of calibration during startup:
https://www.dropbox.com/s/59228a954c94rbw/Osaka01startup.mov?dl=0

And here is a video of me moving to three positions. You can see that the head takes some times before stabilizing a little, but it still moves. (ignore the blinking)
https://www.dropbox.com/s/y6y1w8qqsvo94l4/Osaka01Head.mov?dl=0

And here is the code. It was working fine before:
https://www.dropbox.com/s/2p6ywar5a6hbcg7/icubgazer.cpp?dl=0

julijenv, 2018-08-28:
Hi @kt10aan ,
Could you please check when the robot has finished to calibrate that if you try to move the head frontwards or backwards if the neck cables get loose, if yes please try to find the 4 different tensionner  (2 for each front and back pulleys) to tight them properly.
let me know

kt10aan, 2018-08-29:
@julijenv , when i try to move the head back and forth, it does move but only one or two degrees or so. By the 4 different tensioners, which ones exactly you mean? I mean, which screws.

julijenv, 2018-08-29:
Hi @kt10aan 
the most important things is not to know if the head is moving or not (because of course the backlash of the motors head are not null), the most important thing is to check if when you force the head back and forth backward and frontward if the neck cables got loose - untensed - if  yes and only if yes try to retense the 4 extremities of those 2 cables by retightening them thx to the tensionners.
if you look at the following file:
[Head_V2_Cabling.pdf](https://github.com/robotology/icub-support/files/2331714/Head_V2_Cabling.pdf)
you may understand what I am refering to,
let me know if it helps

kt10aan, 2018-08-29:
From what I can see (its quite cramped in there, they remain tight on their places while I move the head back and forth.

julijenv, 2018-08-29:
Hi @kt10aan , ok so if the mechanical plays are not present at all, I can just conclude that the problem is somewhere else. But from the video I cant really notice a big disturbances when the ball is right in front of the robot it remains quite still in the other hand on the positions on both really left and right it appeared a bit of shaking but what I suspect is that it is because you dont use the torso and by that the head is way too much tilted. Does it make sense to you? in any case if @pattacini  could have a look at the videos he may suggest you some way or another to smooother the movment and reduce the skhaiking.

pattacini, 2018-08-29:
Hi @kt10aan 

The code you're using seems to rely on a quite old infrastructure if I'm not mistaken. I can say that given the calls to the "ancient" directives `YARP_DECLARE_DEVICES` and `YARP_REGISTER_DEVICES`.

Thus, what's the status of your system? Which `yarp`+`icub-main` software are you currently using?
Did you try a clean update? Are all the machines of your cluster aligned? What about the network's health (latencies, delays, drops)?

My first suggestion would be to make a full system update and then launch the gaze controller for doing the following test.

#### Set up the controller's options
```sh
$ yarp rpc /iKinGazeCtrl/rpc
> set Tneck 0.8
> set Teyes 0.4
> set track 0
> bind pitch -20.0 20.0
> bind yaw -40.0 40.0
> bind roll 0.0 0.0
> set sacc 0
```

#### Issue gaze movements
Remeber to wait until the movement has come to a complete stop before yielding the next `look` command:
```sh
> look 3D (-0.5 0.0 0.36)
> look 3D (-0.5 -0.3 0.36)
> look 3D (-0.5 0.3 0.36)
> look 3D (-0.5 0.3 0.0)
> look 3D (-0.5 0.3 0.5)
> look 3D (-0.5 0.0 0.36)
```

How's the behavior now? Still the same?

kt10aan, 2018-09-05:
@pattacini , I did the test and the behavior was perfect.

kt10aan, 2018-09-05:
@pattacini , to answer your previous questions: The YARP + iCub software was the master branch in January 2018. All the machines in the network are aligned (I assume you mean timewise), network has been checked for delays and dropped packets. And a clean update I tried today with the results written above.

pattacini, 2018-09-06:
I've looked over into your snippet and realized that you're enforcing a further unnecessary closed-loop that is most likely causing the problem. Diff the two files in [icub-support#665.zip](https://github.com/robotology/icub-support/files/2356393/icub-support.665.zip) to find out more.

Consider that:
- the Cartesian error is already handled inside the gaze controller;
- since your target is in 2D, don't compute any error signal passing through the 3D, which only adds a bogus projection.

That said, I don't know why/how it was working before.

kt10aan, 2018-09-07:
@pattacini , thanks! But even with taking out the loop you proposed, the behavior remains the same. The only thing that improves it a bit is increasing the CtrlThread period.I am still baffled about the why.

pattacini, 2018-09-07:
Try the following commands then, which corresponds to the mono-pixel looking modality you're employing:
```sh
$ yarp rpc /iKinGazeCtrl/rpc
>> look mono (left 50.0 120.0 2.0)
>> look mono (left 200.0 120.0 2.0)
>> look mono (left 160.0 50.0 2.0)
>> look mono (left 160.0 200.0 2.0)
>> look 3D (-0.5 0.0 0.36)
```

Moreover, when `iKinGazeCtrl` is launched, can you please check whether the intrinsic parameters of the cameras are correctly loaded from file? There are relevant information printed out.

pattacini, 2018-09-07:
If you notice that increasing the thread period does alleviate a bit the situation then the problem might be related by latencies in the control loop, which in turn might be due to several sources.

Anyway, in our setup, we're used to tuning the period to `0.01 s` without any struggle.

pattacini, 2018-09-17:
Closing, feel free to open it again if needed.

kt10aan, 2018-09-21:
Just to report, the latest commands you posted performed normally. 
The output of iKin is:
```
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]Checking if torso part is active ... yes
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /iKinGazeCtrl/head/rpc:o active at tcp://10.0.0.1:10054/
yarp: Port /iKinGazeCtrl/head/command:o active at tcp://10.0.0.1:10055/
yarp: Port /iKinGazeCtrl/head/stateExt:i active at tcp://10.0.0.1:10056/
yarp: Sending output from /iKinGazeCtrl/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /iKinGazeCtrl/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /iKinGazeCtrl/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]Checking if head part is active ... yes
yarp: Port /iKinGazeCtrl/inertial:i active at tcp://10.0.0.1:10057/
yarp: Receiving input from /icub/inertial to /iKinGazeCtrl/inertial:i using tcp
[INFO]Receiving IMU data from /icub/inertial
[INFO]### neck control - requested POSITION mode: IPositionDirect [available] => POSITION mode selected
[INFO]### computed minimum allowed vergence = 0.5 [deg]
[WARNING]/home/icub/.local/share/yarp/robots/iCubOsaka01/icubEyes.ini: aligning matrix for ALIGN_KIN_LEFT not found!
[WARNING]/home/icub/.local/share/yarp/robots/iCubOsaka01/icubEyes.ini: aligning matrix for ALIGN_KIN_RIGHT not found!
[WARNING]tweak.ini: aligning matrix for ALIGN_KIN_LEFT not found!
[WARNING]tweak.ini: aligning matrix for ALIGN_KIN_RIGHT not found!
[INFO]/home/icub/.local/share/yarp/robots/iCubOsaka01/icubEyes.ini: intrinsic parameters for CAMERA_CALIBRATION_LEFT found:
[INFO]w  = 640
[INFO]h  = 480
[INFO]fx = 414.466
[INFO]fy = 414.377
[INFO]cx = 378.193
[INFO]cy = 248.512
[WARNING]tweak.ini: intrinsic parameters for CAMERA_CALIBRATION_LEFT not found!
[INFO]/home/icub/.local/share/yarp/robots/iCubOsaka01/icubEyes.ini: intrinsic parameters for CAMERA_CALIBRATION_RIGHT found:
[INFO]w  = 640
[INFO]h  = 480
[INFO]fx = 417.378
[INFO]fy = 417.59
[INFO]cx = 325.249
[INFO]cy = 232.607
[WARNING]tweak.ini: intrinsic parameters for CAMERA_CALIBRATION_RIGHT not found!
yarp: Port /iKinGazeCtrl/xd:i active at tcp://10.0.0.1:10058/
[INFO]Starting Pseudoinverse Reference Generator at 20 ms
[INFO]Pseudoinverse Reference Generator started successfully
[INFO]Starting Solver at 20 ms
[INFO]Solver started successfully
yarp: Port /iKinGazeCtrl/x:o active at tcp://10.0.0.1:10059/
yarp: Port /iKinGazeCtrl/q:o active at tcp://10.0.0.1:10060/
yarp: Port /iKinGazeCtrl/events:o active at tcp://10.0.0.1:10061/
[INFO]Starting Controller at 10 ms
[INFO]Controller started successfully
yarp: Port /iKinGazeCtrl/mono:i active at tcp://10.0.0.1:10062/
yarp: Port /iKinGazeCtrl/stereo:i active at tcp://10.0.0.1:10063/
yarp: Port /iKinGazeCtrl/angles:i active at tcp://10.0.0.1:10064/
yarp: Port /iKinGazeCtrl/angles:o active at tcp://10.0.0.1:10065/
[INFO]Starting Localizer at 10 ms
[INFO]Localizer started successfully
yarp: Port /iKinGazeCtrl/rpc active at tcp://10.0.0.1:10066/
```

pattacini, 2018-09-21:
Then, there's very little we can do on our side 🤔 

You might consider giving your program a further whirl in order to dump the pixel targets provided to the gaze controller. Just open up a port to stream out those targets and then connect a `yarpdatadumper` to that port.

Afterward, we might use these data to feed the simulator, for example, and find something else.

Just in case, you can also start the gaze controller with the `--debugInfo on` command line option so that you can also dump data from `/iKinGazeCtrl/dbg:o`.

closed 2018-10-11 by pattacini