robotology/mesh-support #666: Issues booting new live image on pc104

pi-q, 2018-06-04:
Hello!

We created a new USB drive with live image for the pc104, following the instructions in http://wiki.icub.org/wiki/The_Linux_on_the_pc104.

Once done, we tested it by booting it through another pc: it boots without issue and we could establish network connection; ssh pc104 works.

However, when we plugged the USB drive into the iCub head, we found that we were unable to reach pc104 (either by ping or ssh).

What should be our next step in figuring out what is the problem? 

At the moment we do not have a VGA cable that can connect to the pc104, but we could hack one following instructions in [http://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf](http://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf). Otherwise we were wondering if we could use a USB hub to connect the USB drive + a monitor (through USB->VGA adaptor that we could buy); is there any reason that this solution would not work?

julijenv, 2018-06-05:
Hi @pi-q , 
i just update the file describe howto make the PC104 to VGA cable you may find some part component to buy to build it. and some pictures

pi-q, 2018-06-05:
We managed to boot from the USB key on another pc (not on the pc104) to check if there were issues. There is an error message when booting:

Error missing /isolinux/splash.png
Press a key to continue

Would it be possible that the PC104 is not booting because it is stuck on this " missing splashscreen" ?

julijenv, 2018-06-05:
Hi @pi-q,
I double check, the update is ok?
just scroll down the pdf file it is 3 pages long now

pi-q, 2018-06-05:
Hi @julijenv,

From here it does not seem so. We followed this link: [https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/](https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/) and there we could find 'Connect_VGA_to_PC104.pdf' and 'Connect_VGA_to_PC104.docx', but both of them are only 2 pages long.

Are we following the wrong link?

mbrunettini, 2018-06-05:
Hi,
the live system writes some info about the network card bus connection the first time it boots on a machine. This implies that if you boot the live on a desktop and then you plug it to a different machine the network may not work (the new network card can be on a different bus.
To fix this you have two options:

1. quick and dirty - open the file `/etc/udev/rules.d/70-presisten-net.rules` and remove any cople of lines like the following: 

`# PCI device 0x8086:0x10f5 (e1000e)

  SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="00:1f:16:0b:47:4d", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth*", NAME="eth0"`

2. clean and easy - burn another USB and FIRST install on the PC104

pi-q, 2018-06-05:
Hi @mbrunettini,

I am sorry, I was entirely not clear in the first message. The FIRST thing we did was **clean and easy** - burn a new USB and FIRST install it on the PC104. At that point, we observed that we could not establish communication with the PC104. 

It is after that, that we tested it by booting it through another pc, and we observed that it boots without issue and we could establish network connection. We confirmed then that we still obtained the same result (no communication) by connecting it on the PC104 a second time.

We could of course burn yet another USB and FIRST install it on the PC104. However we can expect the results to be the same as above.

mbrunettini, 2018-06-05:
I've read above: 

''Once done, we tested it by booting it through another pc: it boots without issue and we could establish network connection; ssh pc104 works.

However, when we plugged the USB drive into the iCub head, we found that we were unable to reach pc104 (either by ping or ssh).''

Then I thought it was that kind of issue.
You can anyway try to burn another USB, it is a low cost try.
if it fails again, you should connect a VGA cable to the pc104 to check what's going on. we can support you in the debugging.

InriaBrice, 2018-06-05:
I have severals question about the VGA cable, first, does the motif of the pins matter ?
[image]
[image]

Second thing, there is numbers on this image, and on a VGA cable too, which should be like that : 
[image]
which seems correspond to the manual
[image]
but if i look on my VGA, i can see numbers written which are like that : 
[image]
So if i take the same motif, pin 15 will be at the top corner left, instead of 5,
And if i take it like that it means that pin 1 is pin 5.

So i'm pretty confused here.

Best regards,

InriaBrice, 2018-06-05:
the output should be a female on the documentation ? and then we add a vga to vga ?

serena-ivaldi, 2018-06-06:
@mbrunettini we burned another flash card and tried it on the pc104 directly, and it still does not boot (or it is unreachable).

BTW @InriaBrice burned the icub live 2017 version from here: http://www.icub.org/download/pc104-images/

julijenv, 2018-06-06:
Hi @InriaBrice ,
the config doesnt really matter, actually you are right the connector which plugs into the screen is a male, but whatever if it is a male or a female connector , the numbering remains the same.
Just follow the instruction of the pinout numbers and u should be ok.

InriaBrice, 2018-06-06:
Hi @julijenv , what is the worst scenario if the pin are not correct ? inverted or what ever

julijenv, 2018-06-06:
usually the worst scenario is when you dont plug the 2x5 pin connector in the correct position, the immediate result is the cable burns.
hope this help.

julijenv, 2018-06-07:
ANy news?
@pi-q  @InriaBrice

pi-q, 2018-06-07:
Hi @mbrunettini, @julijenv,

We should be receiving components for the cable tomorrow, so we hope it will be ready in the morning. Would you be available tomorrow, 2pm, that we can call you then if we need extra help?

InriaBrice, 2018-06-08:
[image]
There is an issue with the GND on pin 10, it seems it's not correct...

InriaBrice, 2018-06-08:
Can you give us the documentation of the pc104 board to check the connection ?
Best regards

julijenv, 2018-06-08:
ooooochhhh!!!
hi @InriaBrice , @serena-ivaldi , @pi-q 
here is the cabling scheme pin tp pin in between 10 pin connector to db15.
ksut follow that.
[db15to10pinCon.xlsx](https://waffleio-direct-uploads-production.s3.amazonaws.com/uploads/564dabdcd8616f15001b0f59/125516c66e82c728ace21e0d46ec96cb76968cf8dff9b64d8752fed665043761260ea62864c7b96aea142f000b0946fd140b0a11fafc8178a6b668778e5f09bec3714d36976345f3e5db55ea43a82a994e8dbcbd8f4ec8ec3a22e2888c4c36161755.xlsx)

julijenv, 2018-06-11:
Hi @InriaBrice , sorry, my bad, cable DB15pin connector pin 13 is attcachd to pin 9 of connector 10pin
[image]

InriaBrice, 2018-06-11:
Dorian and I agree on the same thing : this is bullshit. It's making sparks when we just approach (not even connect) the cable to plug it to the screen. And I did the perfect same connection between the pins. Verified again with the multimeter.

julijenv, 2018-06-12:
Hi @pi-q , @InriaBrice , @dogoepp ,
I triple check again and the last of my pin to pin connection is corerct on 3 of my cables.
here is some pictures that I  took some minutes ago on one of my robot.
[image]
[image]
My colleagues reminded me that one or more I dont remeber exactley the number got burnt not jsut because of 180 degreee OR that the ground of the TV screen is NOT the same as the one of the robot. Could you check both?!?!

InriaBrice, 2018-06-12:
I tryed with a new usb key of 16 GB USB 3.0 compatible usb 2.0, instead of 32 GB key, and it seems to work (i mean i can see the machine and ping his IP).

I opened a terminal in the usb key, then su to be as root user. Then 
bash /utils/linux/makeboot.sh /dev/sdc1 (the device for the usb key).

Here it is what i did when running the script :
[makeboot_output.txt](https://github.com/robotology/icub-support/files/2094233/makeboot_output.txt)

mbrunettini, 2018-06-12:
Just to clarify: please note that `\dev\sdc1` is a **partition** and not a **device**
The corresponding device is `\dev\sdc`
See [this link](https://www.tldp.org/HOWTO/Partition/devices.html)

Then are you able to login to the PC104 now?

mbrunettini, 2018-06-12:
good news!
I updated the wiki, according to your experience with 32 GB drive, thanks.
Can we close the issue?

InriaBrice, 2018-06-12:
Yes thanks for the help

mbrunettini, 2018-06-12:
You're welcome.
Closing issue

closed 2018-06-12 by mbrunettini

serena-ivaldi, 2018-06-18:
If I may add a comment, we can close the issue because we "solved" the PC104 problem, but the cable problem is still not solved. It would be good to find where is the issue on that as well.. otherwise the problem remains pending / unsolved