robotology/mesh-support #667: Left thumb joint oscillation

fbottarel, 2018-06-04:
### Description of the failure
The left thumb oscillates when a position reference is being provided.

### Detailed conditions and logs of the failure
Providing the left thumb proximal joint with a position different than 0 causes the joint to move with no steady state. The range of such oscillation (measured from  `yarpscope` and `yarpmotorgui`) is between 4 and 5 degrees in magnitude. Idling the proximal thumb joint prevents the oscillation from showing up.

julijenv, 2018-06-06:
Hi @fbottarel ,
I rewired joint 7,8 and 9 of the elft hand.
I started the robotInterface and it doesnt move anymore. I noticed by the way that the right thumb where actually moving?

fbottarel, 2018-06-06:
Honestly, I didn't pay too much attention to the right thumb. During startup, the right hand seemed to behave as normal.

closed 2018-06-06 by julijenv