robotology/mesh-support #669: Cables 7,8 and  15 Replacement

HamidMetu, 2018-06-06:
We have an icub robot in our lab. But two of its cables, 7 and 8 was broken and we need to replace them with new ones. I have provided the cables from the icub company. But I do not  know how to replace them.
I would really appreciated it if you help me with these issues by providing me by any documents or videos.
Regards,
Hamid

julijenv, 2018-06-07:
Hi @HamidMetu ,
there is no video tutorial. But there is this document actually with a lot of inside comment on how to replace the shoulder cables.
https://svn.robotology.eu/repos/iCubHardware/trunk/mechanics/public/doc/howTo/tendon replacement guide/cabling manual - head-shoulder-elbow-torso - ENG-ITA.pdf
let us know

julijenv, 2018-06-14:
Hi @HamidMetu ,
Can u tell me if you repaired it, and if yes could you please close the isssue or tell me if I can close it.
thx in advance

HamidMetu, 2018-06-15:
Julien,
Sorry for answering late. Yes, I repaired it. You can close it.
Regards,
Hamid

closed 2018-06-19 by julijenv