robotology/mesh-support #671: Replacement tendons needed

emilio-cartoni, 2018-06-12:
Today our cable 7 in the right shoulder broke but we have no ready replacement (we only have replacement for hand tendons).
I've read in #225 that keeping 3x replacements for all arm tendons (5 xfor cable 7 and 8) is suggested, so we would like to make an order to get 3x of all cables, 5x for cable 8 and 6x for cable 7 (since it just broke).

We might also be interested into head and torso tendons, can we get a quote for them as well?
(we don't move that joints much, so maybe just 1x?)

julijenv, 2018-06-12:
Hi @emilio-cartoni ,
could you please contact andrea.derito@iit.it or send me a direct email at julien.jenvrin@iit.it  with the name  (email fo your superior) in order to start a quotation - I know we have a set of cable for V1 robot and V2 robot, I think it will be fast.
let me know

emilio-cartoni, 2018-06-12:
Hi @julijenv !
I've sent you an email with the request and my supervisor in copy :)

julijenv, 2018-06-12:
hi @emilio-cartoni ,
send me a picture of your robot neck and forearm please

emilio-cartoni, 2018-06-12:
[image]
[image]
[image]
[image]

julijenv, 2019-03-18:
Hi @emilio-cartoni , you received the tendons, AM I right?
if yes, could you close this issue , thx in advacen

closed 2019-08-01 by julijenv