robotology/mesh-support #673: Left arm goes in an uncontrolled state

S-Dafarra, 2018-06-14:
### Description of the failure
The left arm goes in a state which is not down, i.e. the motors are not completely switched off, but yet the joints starting from the wrist prono-supination cannot be controlled and are kind of "compliant".

### Detailed conditions and logs of the failure
[logArmDown_part1.txt](https://github.com/robotology/icub-support/files/2102821/logArmDown_part1.txt)
[logArmDown_part2.txt](https://github.com/robotology/icub-support/files/2102823/logArmDown_part2.txt)

cc @julijenv

S-Dafarra, 2018-06-14:
Also there is a super annoying whistle coming from the hand. If we squeeze the fingers (moving the ``l_hand_fingers`` joint), it stops.

julijenv, 2019-03-20:
Hi @S-Dafarra , still on going this problem?!?!

S-Dafarra, 2019-03-20:
Yes, actually it happens on both arms. For example, yesterday it happened on the right arm. Interestingly we also pressed the fault, but the forearm remained in that strange state.

S-Dafarra, 2019-10-03:
This issue is still occurring.

S-Dafarra, 2019-10-04:
It just happened today, again on the left arm. As far as I know, this is still happening on **both** arms.
Here a more recent log 
[log_icub-head_unresponsiveWrist.txt](https://github.com/robotology/icub-tech-support/files/3690179/log_icub-head_unresponsiveWrist.txt)

After a quick F2F talk with @julijenv we suspect this issue may be related to a board resetting, maybe after a voltage drop, which causes the board to remain in an uncontrolled state with a constant PWM output. 
Maybe @valegagge or @marcoaccame may have some opinion on this.

julijenv, 2019-10-24:
Hi @valegagge ,
if you time to check that and give me your thoughts about it I'd be happy,
thx in advacnce

S-Dafarra, 2019-12-23:
The issue is still occurring.

S-Dafarra, 2020-02-21:
It still happens.

S-Dafarra, 2020-02-25:
After a quick f2f with @julijenv we understood that we need to check whether the board is still pingable when this problem is occurring. This may give us some more hints of what is going on.

S-Dafarra, 2020-04-25:
I am not going to give up on this ``stale-bot``! :joy:

S-Dafarra, 2020-07-17:
Yet another log. BY accident in the end we switched off all the boards by mistake before stopping the ``yarprobotinterface``. Nevertheless, the left wrist was in this status since a while.

[log_icub-head_yarprobotinterface_1705_left_wrist_uncontrollable.txt](https://github.com/robotology/icub-tech-support/files/4938892/log_icub-head_yarprobotinterface_1705_left_wrist_uncontrollable.txt)

S-Dafarra, 2020-09-20:
Still happening

pattacini, 2020-09-20:
Let's try to tackle this with the help of @davidetome

davidetome, 2020-09-23:
Ciao,

I took a look at the logs provided. The fault that causes this behaviour is most likely : 

```
 6450 1461.226705 <ERROR>   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD left_arm-eb24-j4_7 with IP 10.0.1.24  even after 0.200232 seconds: CANNOT PROCEED ANY FURTHER
```

As already pointed out in this [comment](https://github.com/robotology/icub-tech-support/issues/673#issuecomment-538350914), does it seem that the board is resetting for some reasons (a short circuit in a certain position of the arm? a wire not well connected?  A fault on the `mc4plus` board?)

I propose to do some tests tomorrow morning but I need help, maybe we need to uncover the arm, check for connections and maybe replacing the board in the last instance. @julijenv are you available for this?  

@S-Dafarra just a couple of questions:

1. Is the robot available tomorrow morning?
2. Is the failure replicable in some conditions (positions) or it happens randomly?
3. Are the `firmware` of the `boards` updated? If not, can we do that? 
4. Is the sofware running on `laptop/server` and on the `icub-head`updated?

I look forward to your replies,  thanks!

cc @pattacini

pattacini, 2020-09-23:
They're doing many tests these days with Greeny. Let's see if the robot is available.

S-Dafarra, 2020-09-23:
Hi @davidetome 
> * Is the robot available tomorrow morning?

Probably yes. We are planning to use it today full day, but tomorrow it should be free. The robot is still reserved for the iRonCub experiments, but they got postponed of one week. Unfortunately, we weren't able to cancel the booking. I think tomorrow morning is ok.

> * Is the failure replicable in some conditions (positions) or it happens randomly?

If it is repeatable, we did not find how. It happens from time to time, especially under stress (when walking or doing other dynamic demos), even without moving the arms. Once, during a demo, it was happening by simply waving the hand. In general, it seems to happen only when moving. Unfortunately, it seems pretty random.

> * Are the `firmware` of the `boards` updated? If not, can we do that?

I tried to update about two months ago, but if you need to update the boards, feel free to do it. If problems occur, we would need to fix them anyway :wink:  

> 4\. Is the sofware running on `laptop/server` and on the `icub-head`updated?

Right now we are using the latest release on the ``icub-head``. The configuration files are over ``master``. 
Anyways, I remember this issue happening since the first time we started the robot after being assembled.

Even if we are not spotting the actual issue, do you think it may be possible to work it around via sfotware?

davidetome, 2020-09-23:
Ciao @S-Dafarra ,

 Answering your question I think it is not so easy to work it around also because it involves both `FW` and `SW`, especially if the board is resetting as it seems; I mean when `yarprobotinterface` starts configures all the boards and services and if a board goes down and up again it can't be anymore usable until  `yarprobotinterface` restarts.
But on this point @marcoaccame is certainly the most suitable person.

 I spoke with @julijenv, and he will be at CRIS until about 12.00.
He also noticed that the robot is just felt down so is going to check if there are damages and let me know if the robot is still usable. 
If yes, I can come and start trying to reproduce the failure checking for the `EB24` board reset. 

Anyway, can you please advise your colleagues and leave the robot powered off so tomorrow I can bring it to `proto` lab?

cc @pattacini @Fabrizio69

S-Dafarra, 2020-09-23:
> Answering your question I think it is not so easy to work it around also because it involves both `FW` and `SW`, especially if the board is resetting as it seems; I mean when `yarprobotinterface` [...]

I see. We usually reboot also the boards. I may be wrong, but restarting the `yarprobotinterface` may not be enough. I was hoping that with some ``yarpmotorgui`` macumba it was possible to get them back (by triggering the calibration again maybe), but I am just wondering. I totally understand the complexity.

> He also noticed that the robot is just felt down

Yeah, I have some responsibility :grin: 

> let me know if the robot is still usable.

For this I would trust the opinion of @julijenv 

> Anyway, can you please advise your colleagues and leave the robot powered off so tomorrow I can bring it to proto lab?

Yes, I will either do it by myself or ask someone to do it.

davidetome, 2020-09-24:
Juast an update, the robot is not starting due to a problem on the power supply (backpack, some mechanical parts are bended).
@julijenv  isdisassembling the part and going to check

davidetome, 2020-09-24:
Another Update,

The robot now starts again (it was a problem with `BMON` board), @julijenv straightened the bent mechanical part.
We faced another issue with the `EXT FAULT` that was seen pressed by `yarprobotinterface` tough the button was not really pressed.
Davide Gandini checked the wires and cabling fixing it (probably the connector not well inserted after the fall.

Now I'm going to move the `left-arm` for a while using a sequence in `yarpmotorgui` and checking if the board `EB24` restarts

davidetome, 2020-09-24:
Las t update

the robot is running this [sequence](https://github.com/robotology/icub-tech-support/files/5277083/la.posleft_arm.txt) on the `left-arm` for about 2 hours and the board `EB24` is still up and running.

I propose to plan another day (at least) with @julijenv help to do some tests uncovering the arm.

@S-Dafarra you can take back the robot for the moment :+1: 

cc @pattacini

S-Dafarra, 2020-09-24:
> the robot is running this [sequence](https://github.com/robotology/icub-tech-support/files/5277083/la.posleft_arm.txt) on the `left-arm` for about 2 hours and the board `EB24` is still up and running.

Thanks a lot @davidetome. Just to understand, more or less which is the motion of the arm you were trying? Unfortunately, we never understood the root cause. Sometimes it happened also when running the torque-controlled Yoga++, where the arms are moved very little. Also, it happens on both arms.

 In case it happens again while doing some other experiments, can we call you or there is something we can try/save?

davidetome, 2020-09-25:
Ciao @S-Dafarra , I simply cycled some position moving most of the joint of the `left-arm` using the sequence posted above in the `yarpmotorgui` to see if the board `EB24` disappear. 
Honestly, I didn't know that this happens on both the arms,  so this changes a little bit the focus.
 We could consider asking an opinion also to `firmware` guys (@marcoaccame @ale-git )...

cc @julijenv

S-Dafarra, 2020-09-25:
> Honestly, I did not understand that this happens on both the arms, so this changes a little bit the focus.

I probably should have changed the title. I mentioned in a few comments above it was happening on both arms. Unfortunately, I was able to get a usable log only for the left arm. Most of the time it happens after an intensive session, where the log is already saturated since a while.

S-Dafarra, 2021-01-20:
Interestingly, this happened once also on iCub3. So, I guess it is not a problem specific to ``iCubGenova04``.

julijenv, 2021-01-20:
I dont have too much time to deal with this specific issue right now. But at the moment as you define it as non machine specific I would consider it a software issue. and I will let more competent people try to figure out, if it is software related or not before investigated too much time uselessly.
@pattacini

S-Dafarra, 2021-02-16:
I don't know if it is related, but we noticed a similar behavior also for the neck joints. From time to time, the neck joints cannot be controlled anymore and we have to reboot the motors. I suspect that the issue is the same, and I think it has to do with the ``mc4plus`` boards that control all these joints. 

@julijenv @pattacini let me know if you prefer me to change the title of this issue or to open a new one.

cc @DanielePucci

pattacini, 2021-02-16:
>let me know if you prefer me to change the title of this issue or to open a new one.

Title updated! Better off sticking to this issue to keep an eye on the long history.
In this sprint, @davidetome will get in touch w/ you to see how to proceed with this.

S-Dafarra, 2021-02-16:
> > let me know if you prefer me to change the title of this issue or to open a new one.
> 
> Title updated! Better off sticking to this issue to keep an eye on the long history.
> In this sprint, @davidetome will get in touch w/ you to see how to proceed with this.

Ok! Notice that we don't have a lot of time. Starting from next week we will be involved in some shootings outside the lab. It would make our life much easier if we are able to face this during this week. 

At this point, I suspect that the problem is mainly related to firmware as it involves several boards on several robots. The nice thing is that we can use iCub3 for the tests even if we keep the greeny busy.

I don't know if it is related, but I remember a bug on the FT boards that caused them to get stuck because of communication issues: https://github.com/robotology/icub-firmware/issues/102

Is it possible that a similar bug is also present in the ``mc4plus`` firmware?

pattacini, 2021-02-16:
We'll see how we can handle that, although it's difficult to promise to sort out a longstanding problem in a few days.

cc @davidetome @ale-git

davidetome, 2021-02-16:
Ciao @S-Dafarra , if it is feasible me and @ale-git would be on this on Thursday morning(at least) to do some test.. is `iCubGenova09` available?

cc @pattacini @DanielePucci

DanielePucci, 2021-02-16:
I further checked with @S-Dafarra and it is ok on Thursday! Please just book the robot

CC
@robotology/iit-dynamic-interaction-control

DanielePucci, 2021-02-16:
CC @ale-git @davidetome @pattacini

davidetome, 2021-02-18:
> Interestingly, this happened once also on iCub3. So, I guess it is not a problem specific to `iCubGenova04`.

@S-Dafarra since we're going to use iCub3 this morning. on which joint did it happened?

cc @ale-git

pattacini, 2021-02-18:
From the OP:

>joints starting from the wrist prono-supination cannot be controlled and are kind of "compliant"

plus:

>We noticed a similar behavior also for the neck joints. From time to time, the neck joints cannot be controlled anymore and we have to reboot the motors. I suspect that the issue is the same, and I think it has to do with the mc4plus boards that control all these joints.

davidetome, 2021-02-18:
@pattacini , I've read but I was asking if on iCub3 it happened on the neck, on the wrist or on both of them

S-Dafarra, 2021-02-18:
> @pattacini , I've read but I was asking if on iCub3 it happened on the neck, on the wrist or on both of them

On iCub3 we noticed only the wrists for the time being.

S-Dafarra, 2021-02-18:
In any case, let me stress again some of the things we noticed:
- we never managed to reproduce it properly, but lately is happening quite often. It seems to depend on the number of commands we give to the specific part. For example, it seems to happen more often when we use the hands, but it happens also when doing several yogas, when the wrists are not touched at all.
- the boards stop responding. The robot interface starts complaining that it did not have any reply from the board, and, clearly, the values in the motorgui appear to be fixed.
- On the iCubGenova04, it seemed to be happening on the neck when controlling the eyelids for example, so I suppose it is not matter of the power required to the boards.

davidetome, 2021-02-18:
We moved the robot for a while trying (in particular the left forearm) to replicate the fault but it didn't happen.
Anyway looking again at the logs with @ale-git we agree that it should be due to an HW problem since the boards restart.
We agreed with @S-Dafarra to try changing the `mc4plus` controlling the forearms (EB24 and EB27) of the iCubGenova04 and in the meanwhile check wirings both EH and Power.

[image]

cc @pattacini @DanielePucci @maggia80

DanielePucci, 2021-02-18:
If the output of the analysis is to change the `mc4plus`, let us know when you would like to proceed with the substitution

pattacini, 2021-02-18:
Hi @DanielePucci 

The availability of the boards and when to do this replacement needs to be discussed internally w/ @Fabrizio69 and @julijenv first.
Stay tuned.

S-Dafarra, 2021-02-18:
The problem of planning this after the demo, is that the usage of the robot will drop. For iCubGenova04 in particular, the forearms might be removed for a while as well. I am just afraid that a late replacement may give us an answer on whether it worked or not with a lot of delay.

maggia80, 2021-02-18:
I remember there was a timestamp in the eth packets sent by the boards. If the board resets we may see the timestamp restarting from 0.

maggia80, 2021-02-18:
the other issue might be errors in the eth packets. I don't know if it's always on the diagnostic for such errors.

S-Dafarra, 2021-02-18:
> I remember there was a timestamp in the eth packets sent by the boards. If the board resets we may see the timestamp restarting from 0.

But I think that we do not receive anything from those boads, so do we receive those packets?

I remember that we checked with FirmwareUpdater after that happened and the corresponding board was in the IDLE state.

julijenv, 2021-02-19:
Hi everyone:
Just as a matter of sum up, The tests that produce yesterday @davidetome and @ale-git , was already done also on iCubGenova07 with the similar ending, no problem encountered.
I agree that the MC4+ has never been tested by the supplier but I personally doubt about an HW problem - this is why I ask you guys, .
Having talked this morning with @maggia80 , will hope to be able to have more insights from the board timestamp to understand if it resets or not - pehaps @marcoaccame can help us in this. But in the meanwhile we agreed that, asa we will be back from the next event we will try to change one board and keep under strict control to find out if changing the board prevent the forearm to IDLE.
@pattacini , @S-Dafarra , @kouroshD , @DanielePucci

marcoaccame, 2021-02-19:
Hi all, I will have a look at the issue later today.

marcoaccame, 2021-02-19:
here are some sparse preliminary thoughts. i will collect them later on.

In [here](https://github.com/robotology/icub-tech-support/issues/673#issuecomment-779739106), @S-Dafarra wrote:
>I don't know if it is related, but I remember a bug on the FT boards that caused them to get stuck because of communication issues: robotology/icub-firmware#102
Is it possible that a similar bug is also present in the mc4plus firmware?

The `mc4plus` does not use the above mentioned CAN driver. The motion control is done by the `mc4plus` w/out using CAN communication as the actuation terminates on its PWM ports. The communication which is involved is the UDP communication inside the ETH internal network. 

Having said that, thanks @S-Dafarra ... any hypothesis will be investigated.

marcoaccame, 2021-02-19:
Analisis of log of july 2020 as in here.

1. crc error found
   
in the log of july 2020 i have spotted several  CRC errors from board 10.0.1.26 detected in its input port.  that causes the drop of the UDP frame in error w/  several consequences. The first of them is that several otehr boards complain because they detect some missing UDP frames. Further analysis later.  

```
490 1260.241039 <ERROR>  from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 1276s 569m 33u: 
(code 0x06000002, par16 0x0000 par64 0x3000000000000001) 
-> ETH monitor: detected RX CRC error in port specified by par16 
(0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  
Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. 
Number of errors in par64&0xffffffff + .

```

2. the board 10.0.1.24 disappears

the service `EthMonitorPresence` which runs on icub-head detects the absence of one board, the 10.0.1.24, which stays absent for more than 60 seconds.

```
5880 1360.385314 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.24 (left_arm-eb24-j4_7) has been silent for 0.020087 sec (its timeout is 0.02 sec) 
```

This board stays silent for another 60 seconds.

```
6028 1420.389001 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.24 (left_arm-eb24-j4_7) has been silent for another 60.0037 sec, for a total of 60.0238 sec 
```

3. all the ETH boards of the robot disappear

in the log of july 2020 i have spotted that the service `EthMonitorPresence` which runs on icub-head detects the absence of mostly every board inside the robot.

```
6084 1444.767627 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.5 (torso-eb5-j0_2) has been silent for 0.0200102 sec (its timeout is 0.02 sec) 
 
6085 1444.767651 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.6 (left_leg-eb6-j0_3) has been silent for 0.0200291 sec (its timeout is 0.02 sec) 
 
6086 1444.767655 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.7 (left_leg-eb7-j4_5) has been silent for 0.0201232 sec (its timeout is 0.02 sec) 
 
6087 1444.767659 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.27 (right_arm-eb27-j4_7) has been silent for 0.0200164 sec (its timeout is 0.02 sec) 
 
6088 1444.767663 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.29 (right_arm-eb29-j12_15) has been silent for 0.0200527 sec (its timeout is 0.02 sec) 
 
6089 1444.777712 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.1 (left_arm-eb1-j0_3) has been silent for 0.0200806 sec (its timeout is 0.02 sec) 
 
6090 1444.777730 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.3 (right_arm-eb3-j0_3) has been silent for 0.0249074 sec (its timeout is 0.02 sec) 
 
6091 1444.777735 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.8 (right_leg-eb8-j0_3) has been silent for 0.0249684 sec (its timeout is 0.02 sec) 
 
6092 1444.777740 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.9 (right_leg-eb9-j4_5) has been silent for 0.0249648 sec (its timeout is 0.02 sec) 
 
6093 1444.777744 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.11 (right_leg-eb11-skin) has been silent for 0.0249219 sec (its timeout is 0.02 sec) 
 
6094 1444.777748 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.20 (head-eb20-j0_1) has been silent for 0.0248933 sec (its timeout is 0.02 sec) 
 
6095 1444.777751 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.21 (head-eb21-j2_5) has been silent for 0.0202 sec (its timeout is 0.02 sec) 
 
6096 1444.777755 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.22 (face-eb22-j0_1) has been silent for 0.0202057 sec (its timeout is 0.02 sec) 
 
6097 1444.777759 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.25 (left_arm-eb25-j8_11) has been silent for 0.0251045 sec (its timeout is 0.02 sec) 
 
6098 1444.777763 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.26 (left_arm-eb26-j12_15) has been silent for 0.0250769 sec (its timeout is 0.02 sec) 
 
6099 1444.777766 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.28 (right_arm-eb28-j8_11) has been silent for 0.0251465 sec (its timeout is 0.02 sec)
```

board 10.0.1.24 was already lost some time earlier and board 10.0.1.10 is not activated by xml files.

Why all boards? It may be that the ETH goes down, maybe because the cable which goes from icub-head to board 10.0.1.20, the first of the chain.

After that, ... yarprobotineterfce cannot transmit anymore to ETH boards. See messages after.

```
6111 1445.000858 <ERROR> HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.1 after all attempts with id:  ID32 = 0x01000206 -> IND = 2, TAG = eoprot_tag_mc_joint_config_impedance 

etc.
```

marcoaccame, 2021-02-19:
@julijenv, @S-Dafarra: do you have any recent log files?

S-Dafarra, 2021-02-19:
> @julijenv, @S-Dafarra: do you have any recent log files?

Most of the times, the logger gets saturated before. There is a correlation between the logger being saturated and this error because it often happens after we use it for a long time and the total number of messages easily goes above 10000.

> all the ETH boards of the robot disappear

But then, why the rest of the robot is still functional?

DanielePucci, 2021-02-21:
The issue occurred just few minutes before the demo in one of the very bad mikents. Forced to reboot.

[image]

@marcoaccame @maggia80

DanielePucci, 2021-02-21:
At 6.00 p.m. and after 3 hour of continuous shooting only in the afternoon, the problem occurred again, this time on the head. The board was not pingable, and it seems that there was a traffic problem since the cameras were about to get stuck.

[image]

DanielePucci, 2021-02-21:
We tried to reboot the robot (only motors) four times, but whole-body dynamics device was not starting - we checked the logger but nothing particularly strange was happening, no error, although there were many due to other current hw problems so we are nut sure.

So we decided to reboot the head. After reboot though, wbd device kept not streaming. Suddenly, after additional time (~five minutes) the wbd device restarted streaming data.

@marcoaccame @maggia80

DanielePucci, 2021-02-21:
Unfortunately, after we restarted, we were able to run the walking module only once. At the second time, the  walking module blocked when trying to connect to the torso, nor the yarpmotorgui was starting for the same reason. We are not sure if this latter problem (torso stuck) is actually related to this issue. I'm reporting here to detail the sequence of events after one instance of the issue occurrence.

traversaro, 2021-02-21:
> 
> 
> We tried to reboot the robot (only motors) four times, but whole-body dynamics device was not starting - we checked the logger but nothing particularly strange was happening, no error (there were many other errors due to other current hw problems so we are nut sure).
> 
> So we decided to reboot the head. After reboot though, wbd device kept not streaming. Suddenly, after additional time (~five minutes) the wbd device restarted streaming data.
> 
> @marcoaccame @maggia80

FYI this seems an instance of the problem reported and tracked in https://github.com/robotology/icub-tech-support/issues/999 .

DanielePucci, 2021-02-21:
Just for the record, the robot finally crashed without the crane in the very last sequence we were left to shoot before the occurrence of the issue.

We are now attempting fixes with @julijenv and @violadelbono  [image]

We sincerely do not know what happened, probably the wbd being stuck sent wrong data to the walking controller and generated ill posed desired com velocity. We implemented a possible workaround here https://github.com/robotology/walking-controllers/commit/f43685800e48234e9eaf44f3d61c75f614f4d55e

S-Dafarra, 2021-02-22:
Here a recent log. This is actually from yesterday morning, corresponding to the picture of https://github.com/robotology/icub-tech-support/issues/673#issuecomment-782834607, but I managed to upload it only now.

[log_icub-head_yarprobotinterface_6021_boardsStuck.txt](https://github.com/robotology/icub-tech-support/files/6025016/log_icub-head_yarprobotinterface_6021_boardsStuck.txt)

marcoaccame, 2021-02-23:
> Here a recent log. This is actually from yesterday morning, corresponding to the picture of #673 (comment), but I managed to upload it only now.

Hi @S-Dafarra. Did you manage to check the status of board `10.0.1.24`? I mean:
- did it respond to `ping 10.0.1.24`?
- if so, what was its status in `FirmwareUpdater` during the execution of `yarprobotinterace`? `idle` or `running`?

I am asking because in this log file as in the previous one everything starts with board `10.0.1.24` (and only this board) that disappear to the monitoring service run by `yarprobotinterface`. See lines: `eth::EthMonitorPresence: BOARD 10.0.1.24 (left_arm-eb24-j4_7) has been silent for 0.0200384 sec (its timeout is 0.02 sec)` and it does not reappear anymore.
And if the board is not alive (or in `running` mode) than it cannot work properly.

To this board `10.0.1.24` could have happened one of two things:
1. a crash w/out a restart. In such a case it cannot be pinged anymore, nor be seen by `FirmwareUpdater`.
2. a reset of the board, probably due to electrical reasons, w/ a restart. In such a case, the ping will succeed and `FirmwareUpdater` will show the board in `idle` state.

S-Dafarra, 2021-02-23:
> > Here a recent log. This is actually from yesterday morning, corresponding to the picture of #673 (comment), but I managed to upload it only now.
> 
> Hi @S-Dafarra. Did you manage to check the status of board `10.0.1.24`? I mean:
> 
>     * did it respond to `ping 10.0.1.24`?
> 
>     * if so, what was its status in `FirmwareUpdater` during the execution of `yarprobotinterace`? `idle` or `running`?
> 
> 
> I am asking because in this log file as in the previous one everything starts with board `10.0.1.24` (and only this board) that disappears to the monitoring service run by `yarprobotinterface`. See lines: `eth::EthMonitorPresence: BOARD 10.0.1.24 (left_arm-eb24-j4_7) has been silent for 0.0200384 sec (its timeout is 0.02 sec)` and it does not reappear anymore.
> And if the board is not alive (or in `running` mode) then it cannot work properly.
> 
> To this board `10.0.1.24` could have happened one of two things:
> 
>     1. a crash w/out a restart. In such a case it cannot be pinged anymore, nor be seen by `FirmwareUpdater`.
> 
>     2. a reset of the board, probably due to electrical reasons, w/ a restart. In such a case, the ping will succeed and `FirmwareUpdater` will show the board in `idle` state.

Contrarily to what I remembered, immediately after the board was not pingable neither visible from the ``FirmwareUpdater``, so I think we are in case 1

[73 more replies]

closed 2022-01-31 by pattacini