robotology/mesh-support #674: FAILED writng PAGE_ID error while launching yarprobotinterface

hu-yue, 2018-06-15:
### Description of the failure
Sometimes the error message `FAILED writng PAGE_ID` is appearing during the startup of the robot. Apparently, this does not prevent the robot to start successfully. We can directly close the issue, but we wanted to open it just to record this problem.

### Detailed conditions and logs of the failure
~~~
1451	8,498457	WARNING	dcm2rpy() in singularity: choosing one solution among multiple
1373	8,041400	WARNING	embObjMC BOARD  right_leg-eb9-j4_5 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
1449	8,488235	WARNING	dcm2rpy() in singularity: choosing one solution among multiple
1443	8,467971	WARNING	dcm2rpy() in singularity: choosing one solution among multiple
1437	8,457815	WARNING	dcm2rpy() in singularity: choosing one solution among multiple
1419	8,294712	ERROR	FAILED writing  PAGE_ID
1381	8,041595	WARNING	In  right_leg-eb9-j4_5  there isn't 2FOC.AutoCalibration filed. For default it is disabled
1379	8,041546	WARNING	AutoCalibration parameter not found for board right_leg-eb9-j4_5in bottle"2FOC" (HasHallSensor 1 1) (HasTempSensor 0 0) (HasRotorEncoder 1 1) (HasRotorEncoderIndex 0 0) (HasSpeedEncoder 0 0) (RotorIndexOffset 0 0) (MotorPoles 8 8)
1377	8,041498	WARNING	In  right_leg-eb9-j4_5  there isn't 2FOC.Verbose filed. For default it is enabled
1375	8,041450	WARNING	Verbose parameter not found for board right_leg-eb9-j4_5in bottle"2FOC" (HasHallSensor 1 1) (HasTempSensor 0 0) (HasRotorEncoder 1 1) (HasRotorEncoderIndex 0 0) (HasSpeedEncoder 0 0) (RotorIndexOffset 0 0) (MotorPoles 8 8)
~~~

hu-yue, 2018-06-15:
cc @lrapetti @traversaro

hu-yue, 2018-06-15:
Closing, as the issue does not prevent the robot to start.

closed 2018-06-15 by hu-yue

barbalberto, 2018-06-15:
It is related to the configuration of imu bosh on serial port.
In case of error, it retries 5 times, so if you see the error message only once, it means that the second time has succeeded, otherwise you'll see errror `"BoschIMU: set page id 0 failed";`