robotology/mesh-support #675: Almost all boards have errors on detected hole in rx ropframe

hu-yue, 2018-06-15:
### Description of the failure
The first time the robot had the error on hope in rx ropframe on all boards, all the joints stayed controlled but we did not have access to any board. The second time the same errors occured but we could still access the boards.

### Detailed conditions and logs of the failure
We were testing teleoperated walking with the virtualizer, during one of the tests we launched the module and we saw many errors, mainly of the following three types:
1. Detected hole in rx ropframe, on most of boards
```
26574 8289,149815 <ERROR>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 791s 694m 376u: (code 0x00000030, par16 0x0002 par64 0x0000000000015f05) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + .
```
2. CRC error mainly on the arms
```
26553 8289,018255 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 791s 642m 329u: (code 0x06000002, par16 0x0000 par64 0x300000000000000a) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
```
3. Errors similar to this but with different TAGs
```
26590 8289,230793 <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD left_arm-eb25-j8_11 IP 10.0.1.25 and nv ID32 = 0x01000001 -> IND = 0, TAG = eoprot_tag_mc_joint_config
```
```
27809 8322,902937 <WARNING> HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP 10.0.1.8 unsuccessful at attempt num  1 with id:  ID32 = 0x01000011 -> IND = 0, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque 
```

We saw then that the joints were not controllable anymore, the yarpmotorgui went grey as if all boards have been switched off, we tried to ping the boards from `icub-head` and could not ping them. However, the joints were all still controlled, until we switched off the motors.
Log (saturated at the end): 
[log_icub-head_yarprobotinterface_6666_first.txt](https://github.com/robotology/icub-support/files/2106938/log_icub-head_yarprobotinterface_6666_first.txt)

We switched off the robot and could perform a couple of experiments, before the robot again started to have a lot of errors, but his time we could still see the boards and control the joints from the motorgui.
Log: 
[log_icub-head_yarprobotinterface_32361_second.txt](https://github.com/robotology/icub-support/files/2106942/log_icub-head_yarprobotinterface_32361_second.txt)

I might be missing some other type of errors from the log.

hu-yue, 2018-06-15:
@mebbaid @GiulioRomualdi

julijenv, 2018-07-19:
news???, @hu-yue

hu-yue, 2018-07-20:
@julijenv no news, it has not happened again so far. Closing for now.

closed 2018-07-20 by hu-yue