robotology/mesh-support #676: Right shoulder tendon is broken!!

towardthesea, 2018-06-18:
### Description of the failure
As the title says, the right arm of Blackie is malfunction due to the broken tendon, which can be seen by bare eye.

### Detailed conditions and logs of the failure
It happened just after I turn on the `yarprobotinterface`.

towardthesea, 2018-06-19:
Please consider that we will have a demo with **Blackie** tomorrow (20/06/2018).

cc @pattacini @vtikha @Arya07

julijenv, 2018-06-19:
Hi @towardthesea ,
the cable was not broken , needless to say that ist was posisble to see it by the bare eye :D:D:D:D:D.
I rewre everything, you can play now

closed 2018-06-19 by julijenv

towardthesea, 2018-06-19:
Thanks @julijenv  for your decent talent :+1:

towardthesea, 2018-06-19:
Please check the log files with error @julijenv 

[popout_logs.zip](https://github.com/robotology/icub-support/files/2115986/popout_logs.zip)
Thanks :+1: