robotology/mesh-support #677: Ethernet cable drops connection.

ahmadgazar, 2018-06-19:


julijenv, 2018-06-19:
hi @ahmadgazar ,
where is the log??!?!!?!

hu-yue, 2018-06-19:
@julijenv we don't have the log because nothing is shown in the log as the connection just drops. Unlike in #675, we don't have access to the `icub-head`.
We took the extension cable off and attached the original cable directly to the power supply and the connections seems stable. So it might be that there is a problem with the extension cable.

GiulioRomualdi, 2018-06-19:
@julijenv actually the ethernet drop just happened again without the extension (after a couple of hours), so maybe the problem is not the extension. The network is fine because we detached the cable and the robot automatically connected to the wifi.
At this point, maybe it's the cable of the robot, once (about one month ago) we had to change the ethernet cable connector because it was having bad contacts, maybe it is the same or similar issue.

hu-yue, 2018-06-21:
@julijenv during the tests for the demo yesterday (Tuesday), with @GiulioRomualdi and @ahmadgazar we realized that the problem is most probably not the cable, as even with external cables (we tried two differently ones), the connection was lost sometimes. We then performed the demo with the WiFi which seemed to guarantee a more stable connection. So maybe the connector on the robot side is broken or something similar?

julijenv, 2018-06-25:
News? @hu-yue , @GiulioRomualdi

hu-yue, 2018-06-25:
@julijenv, it seems there are not problems anymore, now the router for the wifi is off and the problem seem to be not occurring. Closing this until further issues arise.

closed 2018-06-25 by hu-yue